RPi-Distro/pi-gen #6: resize2fs_once.service startup fails at boot

spiderkeys, 2016-05-06:
Noticed in the boot sequence for a freshly built image that resize2fs_once was failing to get started by systemd. Checked its status and found that the service simply didn't exist. Running raspi-config and expanding the filesystem fixes it and the message doesn't appear again.

XECDesign, 2016-05-06:
might be worth checking stage2 rootfs dir. It should have been added here: https://github.com/RPi-Distro/pi-gen/blob/dev/stage2/01-sys-tweaks/01-run.sh#L5

spiderkeys, 2016-05-06:
Hmm. I do see that there. I'll build a fresh image and try it again, just in case I had accidentally changed something.

iKarith, 2016-09-02:
This issue has been resolved?

XECDesign, 2016-09-07:
Closing for inactivity.

closed 2016-09-07 by XECDesign