RPi-Distro/pi-gen #7: Question: What are steps 01 and 02 in export-image?

spiderkeys, 2016-05-06:
I notice there are two intermediate steps in the export-image stage called set-sources and network. Is there a reason these were added here as opposed to one of the earlier stages? 01-set-sources also has:
```bash
on_chroot sh -e - <<EOF
apt-get update
apt-get -y dist-upgrade
apt-get clean
EOF
```
The update and dist-upgrade seem to be done already in stage0/01-configure-apt/00-run.sh and the clean is done in stage2/03-cleanup/00-run.sh. Wondering if this ends up being redundant.

XECDesign, 2016-05-06:
A dist-upgrade is done in the export stage to be a little more efficient and flexible when testing. There is some logic behind the madness. I may have built an image in the morning, then people found bugs in packages X and that package Z isn't up to date. I can fix the packages, and re-run only the export stage, which is relatively quick. After making sure the bugs are fixed, I can rebuild the image fully. If there are time constraints, I can use it as is, since the only difference is in the image's log files. 

The apt-get clean is a bit of a hacky workaround for a problem. Originally, the apt archives cache directory was mounted from elsewhere to speed up builds and help if there's network trouble or a mirror is down. Since then, I've switched to using apt-cacher-ng, which introduced a problem. If I don't run apt-get clean in the stage that's being exported, when the image is created, the cached packages end up being a part of the image size calculations and also waste time copying across to the image. The right thing to do would be to specifically exclude  the archives directory in du and rsync, but I will leave that for now.

There's a new image coming out which has been going through testing, so if there are any changes before that image comes out, they will be minimal.

spiderkeys, 2016-05-06:
Ah, yea that all makes sense. Good to know, thanks!

closed 2016-05-06 by spiderkeys