RPi-Distro/pi-gen #11: Suggestion: package changes

iKarith, 2016-09-02:
Should `git-core` be removed from `stage3/00-install-packages/00-packages` and replaced with `git` in either stage2 or stage4?  In stage2 it makes -lite a little less lite, and it does fit with the more complete development tools suggested by stage4.

The `gvfs-daemons` package now depends on `udisks2`.  Anything else that uses it in Raspbian requires that or `udisks`.  Currently stage3 installs `udisks`.  Perhaps it is time to update that to `udisks2`?  Doing this will cause a few other packages not to be installed: `hdparm`, `powermgmt-base`, and `cryptsetup-bin`.

XECDesign, 2016-09-04:
Perhaps not for the upcoming release, but I can take a look after the next one.

Any particular reason to use udisks2 instead of udisks if either works?

iKarith, 2016-09-04:
One good reason to prefer udisks2:

```
i A gvfs         Depends  gvfs-daemons (< 1.22.2-1.1~)
i A gvfs-daemons Depends  udisks2
```

You could remove the udisks line entirely and get udisks2 anyway, but 
including it by hand will ensure that it's set to manually installed, 
which is possibly desirable.

Joseph

iKarith, 2017-05-05:
This issue is mostly obsolete, save for the installation of transitional package `git-core` instead of just `git`.

commit 4e5dc43a (2017-05-08): stage4: Install 'git' instead of transitional package 'git-core'

Fixes #11
closed 2017-05-08 by XECDesign