RPi-Distro/pi-gen #12: Suggest: Comments in packages files

iKarith, 2016-09-03:
Currently ##-packages and ##-packages-nr files are just lists of packages to include, one or more per line.  Might it be useful to have comments in these files for later maintenance to indicate *why* packages are there, or to provide some logical grouping?

Here's what I have in mind:

```diff
diff --git a/build.sh b/build.sh
index 5b745e8..84daeda 100755
--- a/build.sh
+++ b/build.sh
@@ -16,7 +16,7 @@ EOF
 		fi
 		if [ -f ${i}-packages-nr ]; then
 			log "Begin ${SUB_STAGE_DIR}/${i}-packages-nr"
-			PACKAGES=`cat $i-packages-nr | tr '\n' ' '`
+			PACKAGES=`sed -e '/[[:space:]]*#.*$/d' < ${i}-packages-nr | tr '\n' ' '`
 			if [ -n "$PACKAGES" ]; then
 				on_chroot sh -e - << EOF
 apt-get install --no-install-recommends -y $PACKAGES
@@ -26,7 +26,7 @@ EOF
 		fi
 		if [ -f ${i}-packages ]; then
 			log "Begin ${SUB_STAGE_DIR}/${i}-packages"
-			PACKAGES=`cat $i-packages | tr '\n' ' '`
+			PACKAGES=`sed -e '/[[:space:]]*#.*$/d' < ${i}-packages | tr '\n' ' '`
 			if [ -n "$PACKAGES" ]; then
 				on_chroot sh -e - << EOF
 apt-get install -y $PACKAGES
```

That will strip out anything starting with # on a line (and leading whitespace).  This will allow stuff like:

```
# X.org compositing effects for LXDE
libgl1-mesa-dri libgles1-mesa libgles2-mesa xcompmgr
```

Scripts can explain what they're doing so that the process is understood by people looking at it later on.  Right now, packages files cannot.  They ought to, just so that they can explain WHY certain packages are being installed and perhaps sort packages into logical groups.

The alternative to doing this would be to have multiple parts of any given stage that install packages, each named for the bundle of packages being installed.  That's semantically quite possible and quite useful, however it's pragmatically a mess simply because Debian has approximately 2^512 packages (only slightly exaggerated) in the database which must be processed every time you run apt-get.

Processing databases takes so long in Debianish systems nowadays that sid's version of aptitude now asks you after finishing an install if you want to quit immediately, without bothering to grind through the package list to regenerate the state graph for the UI before allowing you to do so.  It's ugly and inelegant, but even on fast systems it's become kinda welcome.

The comment removal is fast and orders of magnitude faster than running multiple instances of apt-get when not necessary.

Probably worthwhile to make sed also do the tr stage above, but I didn't bother for comments on the suggestion.  :)

XECDesign, 2016-09-04:
Yeah, absolutely. I thought I already made that modification a while back, but I guess I only thought about it. If you send the PR, I am all for it.

iKarith, 2016-09-05:
Okay, sed and flow control is never pretty, but this appears pretty robust:

```bash
PACKAGES=`sed -e ':a;N;$ !b a' -e 's/[[:space:]]*\(#[^\n]*\)*[[:space:]]/ /g'  < ${i}-packages`
```

Less portable but more readable is possible if you know sed is always GNU sed:

```bash
PACKAGES=`sed -r -e ':a;N;$ !b a' -e 's/[[:space:]]*(#[^\n]*)*[[:space:]]+/ /g'  < ${i}-packages`
```

This was tested with multiple spaces and/or tabs between items, comments on lines by themselves or at the end of lines, and blank lines.  It produces a single-space-delimited list of packages.

I break the script into two expressions so that it's somewhat readable.  The two expressions can be combined with a simple semicolon.

iKarith, 2016-09-15:
Implemented as of abc3e45, closing.

closed 2016-09-15 by iKarith