RPi-Distro/pi-gen #13: Suggestion to build.sh

ssfdre38, 2016-09-05:
This script adds a check to make sure the user is under root to run the script and adds all dependencies before starting the scripting process and if IMG_NAME is not set, will echo how to set IMG_NAME in the command.
[build.txt](https://github.com/RPi-Distro/pi-gen/files/454448/build.txt)

ssfdre38, 2016-09-05:
This one i changed some stuff around and set it up where it will check for root access first, Makes sure the IMG_NAME is set, and updates the apt list and adds the dependencies before it starts so the system is set up just right for the build.
[build.txt](https://github.com/RPi-Distro/pi-gen/files/454471/build.txt)

XECDesign, 2016-09-05:
I don't know if it makes sense to check the dependencies every time and that apt-get line is a little expensive. Although I only support Mint, I don't want to add lines which I know won't work on non-apt based distros.

We could do something like this:
``` bash
for DEP in quilt kpartx realpath qemu-arm-static debootstrap zerofree pxz; do
  if ! command -v $DEP > /dev/null; then
    MISSING_DEPS="$DEP $MISSING_DEPS"
  fi
done

if [ -n "$MISSING_DEPS" ]; then
  echo "Could not find the following dependencies: $MISSING_DEPS"
  exit 1
fi
```
The problem with that is that the user won't know that qemu-arm-static comes from qemu-user-static, but it's better than just failing later down the line.

iKarith, 2016-09-06:
If you use `pkgname:cmd` format, you can split that in a for loop to test for commands' existence and return a list of necessary packages if not found.  That would allow you to `apt-get` only if the check for missing dependencies finds any.

Suggest `hash` instead of `command -v`.  In this instance it's probably not necessary, but it's a good habit since the former actually looks for the file, rather than assuming it is or isn't there.

The dependency list should probably also include `zip`.  Most people will already have that installed, y'know, unless they don't.  Discovering you forgot to install it on a new system after 74 minutes of watching all the stages finish wasn't amusing.  ;)

ssfdre38, 2016-09-06:
yea i was using raspbian lite that i built myself using the script and when i did my test overnight i saw that i needed zip at the very end. But yea there should be a check for the packages first before starting so that way the base system is set up for the build.

ssfdre38, 2016-09-06:
So i was trying some commands and right now i have it set up where it will check for dpkg and if it is there will not try and install it other if not there it will install if package is not install.

[build.txt](https://github.com/RPi-Distro/pi-gen/files/456114/build.txt)

iKarith, 2016-09-06:
A flaw in your script: IMG_NAME may be set in config, and in fact that's what config is used to set.

Joseph
Sent via mobile

ssfdre38, 2016-09-06:
its not my script and its just there if the IMG_NAME isnt set already

iKarith, 2016-09-06:
You don't understand. It sources a file named config if found. You modified the script to error out before it ever loads the file where you're supposed to set it.

Joseph
Sent via mobile

XECDesign, 2016-09-06:
Although I haven't gone out of my way to support anything other than Mint, I don't want to assume the presence of apt and dpkg or automatically install anything on the user's system.

ssfdre38, 2016-09-06:
I know there is a way to have it check the system first and then check for packages i saw it on a script one time i just need to find that again

XECDesign, 2016-09-06:
I don't think this script needs to be idiot proofed too much. Let the user know the dependencies are missing and let them worry about how to install them on their system. @iKarith's approach of using hash, then determining the source package would be ideal.

iKarith, 2016-09-06:
I'll submit a PR that does this in a bit. I'm off hunting a machine that doesn't tak 147 minutes to run the thing end to end. ;)

Joseph
Sent via mobile

iKarith, 2016-09-07:
This is where I'm headed with it:

```diff
diff --git a/build.sh b/build.sh
index 5b745e8..a9239c3 100755
--- a/build.sh
+++ b/build.sh
@@ -1,5 +1,21 @@
 #!/bin/bash -e
 
+dep_check()
+{
+	for dep in "$@"; do
+		if ! hash ${dep%:*} 2>/dev/null; then
+			missing="${missing:+$missing }${dep#*:}"
+		fi
+	done
+
+	if [[ "$missing" ]]; then
+		echo "missing: $missing"
+		false
+	fi
+}
+
+dep_check quilt kpartx qemu-arm-static:qemu-user-static debootstrap zerofree pxz zip foo:bar frotz
+
 run_sub_stage()
 {
 	log "Begin ${SUB_STAGE_DIR}"
```

I want to clean that up a bit and maybe with decent output and whatnot, make it so dependencies are stored in a file like packages files are now, etc.  I'm also not sure if that really needs to live in build.sh or can be sourced from another file.  Preferences?

iKarith, 2016-09-07:
This I'm happy with:

```diff
diff --git a/README.md b/README.md
index 9ff754f..d6c8510 100644
--- a/README.md
+++ b/README.md
@@ -5,7 +5,7 @@
 
 #Dependencies
 
-`quilt kpartx realpath qemu-user-static debootstrap zerofree pxz`
+`quilt kpartx realpath qemu-user-static debootstrap zerofree pxz zip`
 
 #Config
 
diff --git a/build.sh b/build.sh
index 5b745e8..d78fc20 100755
--- a/build.sh
+++ b/build.sh
@@ -156,6 +156,10 @@ export QUILT_NO_DIFF_TIMESTAMPS=1
 export QUILT_REFRESH_ARGS="-p ab"
 
 source ${SCRIPT_DIR}/common
+source ${SCRIPT_DIR}/dependencies_check
+
+
+dependencies_check ${BASE_DIR}/depends
 
 mkdir -p ${WORK_DIR}
 log "Begin ${BASE_DIR}"
diff --git a/depends b/depends
new file mode 100644
index 0000000..bf8712d
--- /dev/null
+++ b/depends
@@ -0,0 +1,5 @@
+quilt
+qemu-arm-static:qemu-user-static
+debootstrap
+kpartx zerofree
+pxz zip
diff --git a/scripts/dependencies_check b/scripts/dependencies_check
new file mode 100644
index 0000000..269ebd0
--- /dev/null
+++ b/scripts/dependencies_check
@@ -0,0 +1,29 @@
+# dependencies_check
+# $@	Dependnecy files to check
+#
+# Each dependency is in the form of a tool to test for, optionally followed by
+# a : and the name of a package if the package on a Debian-ish system is not
+# named for the tool (i.e., qemu-user-static).
+dependencies_check()
+{
+	local depfile deps missing
+
+	for depfile in "$@"; do
+		if [[ -e "$depfile" ]]; then
+			deps=`cat ${BASE_DIR}/depends | tr '\n' ' '`
+		fi
+		for dep in $deps; do
+			if ! hash ${dep%:*} 2>/dev/null; then
+				missing="${missing:+$missing }${dep#*:}"
+			fi
+		done
+	done
+
+	if [[ "$missing" ]]; then
+		echo "Reqired dependencies not installed"
+		echo
+		echo "This can be resolved on Debian/Raspbian systems by installing:"
+		echo "$missing"
+		false
+	fi
+}
```

@XECDesign this doesn't have my pkg_comments PR applied to it but uses the same mechanism, so if you're ready to merge that I can update this to use the same mechanism.  Also, a side question for future direction's sake: Is this tool really only ever intended for use on Raspbian itself, or is it something that may be used to generate other distributions for the Pi and possibly similar sorts of integrated devices?  I made the dependency filename come from outside first because that seems to belong in build.sh rather than a dependency, and also because if we decide to move Raspbian stages into a subdir (possibly named by IMG_NAME?), a dependency file in that subdir can also be passed in.

Room for improvement: qemu-user-static is only needed when not armhf.  Currently I don't have a way to specify an arch-dependency.  I could extend the syntax for that I suppose.

XECDesign, 2016-09-07:
Looks good.

While writing the tool, I was trying to keep it generic in case we ever need to build x86, stretch, ubuntu or debian images. Officially, it only supports what we release (Raspbian), but I am not opposed to changes which improve portability without sacrificing simplicity.

For different releases, the plan was to have separate branches rather than separate subdirectories. Originally, there was a 'base' branch, which didn't have any of the stages. Raspbian stages were an example for people who wanted to create their own image, or a starting point for people who wanted to customise the official image.

I have been playing around with Debian's Live CD build tool and would like to take some pages from their book (not any time soon). They keep the configuration entirely separate by generated using a command or loading it from a remote source.

iKarith, 2016-09-07:
I looked at Debian's live-build a little. Very flexible, but suffers somewhat from complexity sometimes for the sake of it. I asked about generic builds because I _do_ need to build Raspple II for i686 and amd64 as well.

Separating config from build tool works well enough I suppose, but I'd argue keeping it simple would be best for contributors to join in.

Joseph
Sent via mobile

ssfdre38, 2016-09-11:
well i was looking at the build and set up a config file wit all the arguments added and the build.sh didnt want to read from the config file at all

iKarith, 2016-09-12:
Where was config? it should be in your current directory—not the directory of build.sh.

Joseph
Sent via mobile

ssfdre38, 2016-09-12:
well i had it in pi-gen folder but i was also looking in the build code and there wasn't anything for the config

iKarith, 2016-09-13:
At the moment, config is only useful for image name and apt proxy really. So it almost seems like there's no point in having it. Except that there are plans to support having more than one set of build scripts and the workings of Debian's live CD tools are a good model. There was also discussion of the possibility of multi-arch support in the scripts—after all the Pi 3 can do arm64 and I personally also need to build i686 for Rasplle II as well as arm6hf. All good reasons to have a config file. More useful might be a patch for processing an argument to specify which config file to load.

Anyway, config is now described some in README.md.  More work is needed there, and I'll get to it when I'm finished wrangling LUKS on my Debian box. There's enough potential in pi-gen and I have my own motives for improving the tool chain that I'm likely to be a semi-regular contributor for a time if the RPF/Raspbian folks will have me.

Joseph
Sent via mobile

ssfdre38, 2016-09-13:
well that is what i tested in the my testing environment and build.sh doesnt know it there nor knows to read it cause i had it set to do IMG_NAME and APT_PROXY and it failed on both arguments

iKarith, 2016-09-14:
It works something like this:

```
tjcarter@rem:~/Source/pi-gen-test$ echo 'IMG_NAME="Raspbian"' > config
tjcarter@rem:~/Source/pi-gen-test$ cat config 
IMG_NAME="Raspbian"
tjcarter@rem:~/Source/pi-gen-test$ time sudo ../RPi-Distro/pi-gen/build.sh 
[04:10:01] Begin /home/tjcarter/Source/RPi-Distro/pi-gen
[04:10:01] Begin /home/tjcarter/Source/RPi-Distro/pi-gen/stage0
[04:10:01] Begin /home/tjcarter/Source/RPi-Distro/pi-gen/stage0/prerun.sh
I: Running command: debootstrap --arch armhf --foreign --components=main,contrib,non-free --no-check-gpg jessie /home/tjcarter/Source/RPi-Distro/pi-gen/work/2016-09-14-Raspbian/stage0/rootfs http://mirrordirector.raspbian.org/raspbian/
[... GRIND GRIND GRIND GRIND ...]
[05:13:41] Begin /home/tjcarter/Source/RPi-Distro/pi-gen/export-noobs
[05:13:41] Begin /home/tjcarter/Source/RPi-Distro/pi-gen/export-noobs/prerun.sh
umount: /: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)

real    72m34.969s
user    70m47.640s
sys     2m21.440s
```

You can put `config` in the directory with `build.sh`, but only if you run it from that directory, which I am not doing here.  The use of `time` is totally optional--I just wanted to see how long it'd take on a "warm" run since I hadn't timed any of those yet.  My intention is to get the time down to under 70 minutes on a cold run if I can--but that's neither here nor there.

Okay, so you run it there, and where's your output?  The clue is that I said it was a warm run--it's in the directory with `build.sh`!  \*sigh\*  Obviously, you're meant to the pi-gen directory and run it from there, but in that case checking for config in $PWD is nonsensical.  I've got thoughts about that, but they'll need to take a back seat to me doing some work using it as-is for now.  If one of the RPF and Raspbian people don't improve that before I get back to it, I'll propose something and write up the patch if they like it.

XECDesign, 2016-10-07:
@ssfdre38 would you say this one can be closed?

closed 2016-10-09 by XECDesign