RPi-Distro/pi-gen #17: dependencies_check fail

ssfdre38, 2016-09-14:
so i was testing the new build with sudo IMG_NAME="raspbian" ./build.sh and there is an error tossed from the dependencies_check file 
/home/ssfdre38/pi-gen/scripts/dependencies_check: line 27: unexpected EOF while looking for matching `"'
Line 27 is echo "$missing"

hadmack, 2016-09-15:
I have the same problem, seems to be due to line 13 nested quotes for sed:

    deps="$(sed -f "${SCRIPT_DIR}/remove-comments.sed" < ${BASE_DIR}/depends"
  
but I don't know how to fix this.  It clearly must work on some system but not with my bash version.

XECDesign, 2016-09-15:
Sorry, should've tested before merging that PR. Seems to just be a missing ')'.

iKarith, 2016-09-15:
I must've accidentally deleted it just before committing, sorry.  :(

XECDesign, 2016-09-15:
No worries, happens to everyone.

iKarith, 2016-09-15:
(This got reopened by the comments, it should be re-closed.)

XECDesign, 2016-09-15:
I was waiting for @ssfdre38 to close it or acknowledge that it's no longer an issue for him. Github didn't actually close this to begin with since the 'issue' in the commit message breaks that mechanism.

closed 2016-09-15 by ssfdre38