RPi-Distro/pi-gen #18: Suggestion: Use sudoers.d instead of patching /etc/sudoers

iKarith, 2016-09-15:
Currently stage2/01-sys-tweaks applies a patch to `/etc/sudoers`, which is a Debian conffile.  If at all possible, we shouldn't be modifying Debian conffiles.  The reason for that is that should /etc/sudoers ever change in Debian, the user is going to be asked whether they want to keep "their" local changes or adopt Debian's.

Instead, I propose the solution I use on my own systems:

```diff
diff --git a/stage2/01-sys-tweaks/00-patches/00-sudoers.diff b/stage2/01-sys-tweaks/00-patches/00-sudoers.diff
deleted file mode 100644
index 3fda3d3..0000000
--- a/stage2/01-sys-tweaks/00-patches/00-sudoers.diff
+++ /dev/null
@@ -1,10 +0,0 @@
-Index: jessie-stage2/rootfs/etc/sudoers
-===================================================================
---- jessie-stage2.orig/rootfs/etc/sudoers
-+++ jessie-stage2/rootfs/etc/sudoers
-@@ -25,3 +25,5 @@ root	ALL=(ALL:ALL) ALL
- # See sudoers(5) for more information on "#include" directives:
- 
- #includedir /etc/sudoers.d
-+
-+pi ALL=(ALL) NOPASSWD: ALL
diff --git a/stage2/01-sys-tweaks/00-patches/series b/stage2/01-sys-tweaks/00-patches/series
index b1381c8..5cb4c55 100644
--- a/stage2/01-sys-tweaks/00-patches/series
+++ b/stage2/01-sys-tweaks/00-patches/series
@@ -1,4 +1,3 @@
-00-sudoers.diff
 01-useradd.diff
 02-swap.diff
 03-console-setup.diff
diff --git a/stage2/01-sys-tweaks/01-run.sh b/stage2/01-sys-tweaks/01-run.sh
index 9356ce6..7a9759c 100755
--- a/stage2/01-sys-tweaks/01-run.sh
+++ b/stage2/01-sys-tweaks/01-run.sh
@@ -8,6 +8,8 @@ install -d						${ROOTFS_DIR}/etc/systemd/system/rc-local.service.d
 install -m 644 files/ttyoutput.conf			${ROOTFS_DIR}/etc/systemd/system/rc-local.service.d/
 
 install -m 644 files/50raspi				${ROOTFS_DIR}/etc/apt/apt.conf.d/
+install -m 440 files/group_sudo				${ROOTFS_DIR}/etc/sudoers.d
+
 
 
 on_chroot sh -e - <<EOF
@@ -25,7 +27,7 @@ on_chroot sh -e - << \EOF
 for GRP in input spi i2c gpio; do
 	groupadd -f -r $GRP
 done
-for GRP in adm dialout cdrom audio users sudo video games plugdev input gpio spi i2c netdev; do
+for GRP in adm dialout cdrom audio users sudo video games plugdev input gpio spi i2c netdev sudo; do
   adduser pi $GRP
 done
 EOF
diff --git a/stage2/01-sys-tweaks/files/group_sudo b/stage2/01-sys-tweaks/files/group_sudo
new file mode 100644
index 0000000..44f8752
--- /dev/null
+++ b/stage2/01-sys-tweaks/files/group_sudo
@@ -0,0 +1,2 @@
+# Allow members of group sudo to execute any command without a password
+%sudo	ALL=(ALL:ALL) NOPASSWD: ALL
```

The file /etc/sudoers.d/README explains the rules for this.

NB: This patch does provide a small functionality change.  Instead of giving the pi user explicitly unlimited access to use sudo, it adds pi to the group sudo and then overrides the rule in /etc/sudoers to make that access passwordless.

You _could_ put the group_sudo file into one of the Raspbian mods packages, but that might change behavior for existing systems.  I'd suggest doing it only on new systems as I've done here.  In any event, if you opt to add it to a package instead of doing it here as I've done, mark it as a conffile so that user changes to the file will be preserved unless they choose not to.

iKarith, 2016-09-15:
Oh, yes:

```
tjcarter@rem:~$ id
uid=1000(tjcarter) gid=1000(tjcarter) groups=1000(tjcarter),24(cdrom),25(floppy),27(sudo),29(audio),30(dip),44(video),46(plugdev),107(netdev),112(scanner),118(lpadmin)
tjcarter@rem:~$ lsb_release -d
Description:    Debian GNU/Linux unstable (sid)
tjcarter@rem:~$ cat /etc/sudoers.d/group_sudo
cat: /etc/sudoers.d/group_sudo: Permission denied
tjcarter@rem:~$ sudo -K
tjcarter@rem:~$ sudo cat /etc/sudoers.d/group_sudo
# Allow members of group sudo to execute any command without a password
%sudo   ALL=(ALL:ALL) NOPASSWD: ALL
```

And that's why I did it this way.  Raspple II supports more than Raspberry Pi and we don't use pi as the user on non-Raspbian installations.  That'll work on any system using Debian's default /etc/sudoers file.

XECDesign, 2016-09-15:
The plan is to move changes like this out into raspberrypi-sys-mods in a way similar to this: https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/ac9a70277d8ad94acc9e34f51ce80ae8333ad566

The problem with having these changes in pi-gen is that the files are impossible to update automatically on existing systems in a sane way.

iKarith, 2016-09-15:
The reason I didn't suggest doing it that way instead was that changing existing systems may change behavior.  I suppose you could work around that by giving the pi user special treatment rather than using the group, which would have the effect of at most restoring Raspbian defaults out from under the user.  Which is fine--if you're using a Pi for anything that something like that might matter (Rails cluster node, etc.) you should be using apt-listchanges anyway.  I'd suggest that would a good time to use the group sudo change anyway--and I personally always prefer general solutions to specific ones.

If you'd like to leave this open pending that change being done in raspberrypi-sys-mods, that's fine, else closing it is also fine.  (There's a website I used at one point for moving a ticket from one project to another, but I'd have to dig for it now.)

I'm using this patch for RasppleII/pi-gen, but that's why I've been doing PRs out of iKarith/pi-gen.  The former may eventually diverge from your tree significantly, but at least for now we seem to be reading the same book, if perhaps not exactly the same page.  :)

XECDesign, 2016-09-15:
The behaviour would be identical. When updating raspberrypi-sys-mods, install the new conffile, if sudoers has the old line, take it out, otherwise comment out the line in the new conffile. That should preserve old behaviour regardless of the existing setup, unless I've missed something.

I would be reluctant to give nopasswd to all users in sudo group, since that does change behaviour and people may find that their installs suddenly become a little less secure without notice.

iKarith, 2016-09-15:
Also reasonable.  In which case I won't worry about `/etc/sudoers.d/group_sudo` clashing with your changes on the Pi, which simply means it'll probably become a stage 5 addition in the end, which works for my end as well.

XECDesign, 2016-10-04:
Leaving this as is for now.

closed 2016-10-04 by XECDesign