RPi-Distro/pi-gen #19: export-noobs failure if you use /dev/mapper

iKarith, 2016-09-16:
I posted part of this in my run of pi-gen in #13, and I've noticed I always get the same result:

```
[05:13:41] Begin /home/tjcarter/Source/RPi-Distro/pi-gen/export-noobs
[05:13:41] Begin /home/tjcarter/Source/RPi-Distro/pi-gen/export-noobs/prerun.sh
umount: /: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)

real    72m34.969s
user    70m47.640s
sys     2m21.440s
```

Note this was my RPi-Distro tree, which has no patches.

You'll note that there is no `End /home/tjcarter/Source/RPi-Distro/pi-gen/export-noobs/prerun.sh` to match, and it ends with umount returning an error.  (An error in any stage causes pi-gen to bail, leaving filesystems mounted and other issues, but that's a slightly more complex issue.)  The prerun.sh is failing in its final line to `unmount_image ${IMG_FILE}`.  At first glance, `IMG_FILE` was unmounted, but a second look spots the `kpartx` call which would've placed the image back in use.

Again, this is a problem I see in RPi-Distro's tree, however I'm going to use my own tree to try and diagnose the problem because it has a completely populated work/ from today, resulting in a 15 minute build instead of a 70-90 minute build depending on disk caching.

My debug procedure mostly consists of running build.sh with the following patch:

```diff
diff --git a/export-noobs/prerun.sh b/export-noobs/prerun.sh
index a531be6..bde31b7 100755
--- a/export-noobs/prerun.sh
+++ b/export-noobs/prerun.sh
@@ -1,4 +1,4 @@
-#!/bin/bash -e
+#!/bin/bash -ex
 
 IMG_FILE="${STAGE_WORK_DIR}/${IMG_DATE}-${IMG_NAME}${IMG_SUFFIX}.img"
 NOOBS_DIR="${STAGE_WORK_DIR}/${IMG_DATE}-${IMG_NAME}${IMG_SUFFIX}"
```

Relevant output follows:

```
[00:35:45] End /home/tjcarter/Source/RasppleII/pi-gen/stage4
[00:35:45] Begin /home/tjcarter/Source/RasppleII/pi-gen/export-image
[00:35:45] Begin /home/tjcarter/Source/RasppleII/pi-gen/export-image/prerun.sh
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
mke2fs 1.43.3 (04-Sep-2016)
mount: /dev/mapper/loop1p2 mounted on /home/tjcarter/Source/RasppleII/pi-gen/work/2016-09-16-RasppleII/export-image/rootfs.
mount: /dev/mapper/loop1p1 mounted on /home/tjcarter/Source/RasppleII/pi-gen/work/2016-09-16-RasppleII/export-image/rootfs/boot.
[00:35:57] End /home/tjcarter/Source/RasppleII/pi-gen/export-image/prerun.sh
[00:35:57] Begin /home/tjcarter/Source/RasppleII/pi-gen/export-image/00-allow-rerun
[00:35:57] Begin /home/tjcarter/Source/RasppleII/pi-gen/export-image/00-allow-rerun/00-run.sh
[00:35:57] End /home/tjcarter/Source/RasppleII/pi-gen/export-image/00-allow-rerun/00-run.sh
[00:35:57] End /home/tjcarter/Source/RasppleII/pi-gen/export-image/00-allow-rerun
[00:35:57] Begin /home/tjcarter/Source/RasppleII/pi-gen/export-image/01-set-sources
[00:35:57] Begin /home/tjcarter/Source/RasppleII/pi-gen/export-image/01-set-sources/00-patches
Converting meta-data to version 2
Applying patch /home/tjcarter/Source/RasppleII/pi-gen/export-image/01-set-sources/00-patches/0-sources.diff
patching file rootfs/etc/apt/sources.list.d/raspi.list

Now at patch /home/tjcarter/Source/RasppleII/pi-gen/export-image/01-set-sources/00-patches/0-sources.diff
[00:35:57] End /home/tjcarter/Source/RasppleII/pi-gen/export-image/01-set-sources/00-patches
[00:35:57] Begin /home/tjcarter/Source/RasppleII/pi-gen/export-image/01-set-sources/01-run.sh
Hit http://archive.raspberrypi.org jessie InRelease
Get:1 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]
Hit http://archive.raspberrypi.org jessie/main armhf Packages 
Get:2 http://mirrordirector.raspbian.org jessie/main armhf Packages [8981 kB]
Hit http://archive.raspberrypi.org jessie/ui armhf Packages           
Ign http://archive.raspberrypi.org jessie/main Translation-en          
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Get:3 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [37.5 kB]          
Get:4 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [70.3 kB]         
Get:5 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1356 B]               
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en                      
[... 191 lines cut ...]
+ DIR=/dev/mapper
+ mount
+ grep -q /dev/mapper
++ mount
++ grep /dev/mapper
++ cut -f 3 -d ' '
++ sort -r
+ local 'LOCS=/home/tjcarter/Source/RasppleII/pi-gen/work/2016-09-16-RasppleII/export-noobs/rootfs
/home/tjcarter/Source/RasppleII/pi-gen/work/2016-09-16-RasppleII/export-noobs/bootfs/'
+ for loc in '$LOCS'
+ umount /home/tjcarter/Source/RasppleII/pi-gen/work/2016-09-16-RasppleII/export-noobs/rootfs
+ for loc in '$LOCS'
+ umount /home/tjcarter/Source/RasppleII/pi-gen/work/2016-09-16-RasppleII/export-noobs/bootfs
+ for loc in '$LOCS'
+ umount /
umount: /: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)

real    14m32.352s
user    35m18.548s
sys     0m33.020s
```

As you can see, unmount_image is doing all kinds of not intended things there.

XECDesign, 2016-09-16:
Not sure that makes sense. `LOCS` looks like it's set correctly, but instead of the two locations in there, the for block seems to have an additional '/' element.

iKarith, 2016-09-16:
I suspect that's because / is a LUKS filesystem:

```
tjcarter@rem:~$ mount | grep mapper
/dev/mapper/root_crypt on / type ext4 (rw,relatime,errors=remount-ro,data=ordered)
```

I suspect we need to make changes to the function to protect it from trying to unmount filesystems that are not related to the image file directly.

XECDesign, 2016-09-16:
Yeah, reading over that part of the script and it could certainly use some work.

iKarith, 2016-09-16:
Yeah, I kind of wonder why it needs to use an image file at all.  You build the stage directories in a chroot, then put the chroot in an image file.  Just tar up the contents of /boot on one side, and exclude them on the other.  Trying to get a better look at the problem, I've updated the debugging patch (which requires minor editing to use without [my open PR](https://github.com/RPi-Distro/pi-gen/pull/20) for those playing along at home):

```diff
diff --git a/export-noobs/prerun.sh b/export-noobs/prerun.sh
index a531be6..bde31b7 100755
--- a/export-noobs/prerun.sh
+++ b/export-noobs/prerun.sh
@@ -1,4 +1,4 @@
-#!/bin/bash -e
+#!/bin/bash -ex
 
 IMG_FILE="${STAGE_WORK_DIR}/${IMG_DATE}-${IMG_NAME}${IMG_SUFFIX}.img"
 NOOBS_DIR="${STAGE_WORK_DIR}/${IMG_DATE}-${IMG_NAME}${IMG_SUFFIX}"
diff --git a/scripts/common.sh b/scripts/common.sh
index 0d2e837..8c03125 100644
--- a/scripts/common.sh
+++ b/scripts/common.sh
@@ -41,7 +41,12 @@ unmount(){
 	fi
 
 	while mount | grep -q $DIR; do
+		echo "unmount: Debug: \"mount | grep $DIR\""
+		mount | grep $DIR
 		local LOCS=`mount | grep $DIR | cut -f 3 -d ' ' | sort -r`
+		echo "unmount: Debug: Start of LOCS"
+		echo "$LOCS" | sed -e 's/^/"/;s/$/"/'
+		echo "unmount: Debug: End of LOCS"
 		for loc in $LOCS; do
 			umount $loc
 		done
```

It creates the following relevant output:

```
++ losetup -j /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/2016-09-16-Raspbian.img
++ cut -f1 -d:
+ local LOOP_DEV=/dev/loop1
+ '[' -n /dev/loop1 ']'
++ mount
+++ basename /dev/loop1
+++ head -n 1
+++ cut -f 3 -d ' '
++ grep loop1
+ local 'MOUNTED_DIR=/dev/mapper/loop1p2 on /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/rootfs type ext4 (rw,relatime,data=ordered)
/dev/mapper/loop1p1 on /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/bootfs type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)'
+ '[' -n '/dev/mapper/loop1p2 on /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/rootfs type ext4 (rw,relatime,data=ordered)
/dev/mapper/loop1p1 on /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/bootfs type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)' ']'
++ dirname /dev/mapper/loop1p2 on /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/rootfs type ext4 '(rw,relatime,data=ordered)' /dev/mapper/loop1p1 on /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/bootfs type vfat '(rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)'
+ unmount /dev/mapper . /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs . . . /dev/mapper . /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs . . .
+ '[' -z /dev/mapper ']'
+ DIR=/dev/mapper
+ mount
+ grep -q /dev/mapper
+ echo 'unmount: Debug: "mount | grep /dev/mapper"'
unmount: Debug: "mount | grep /dev/mapper"
+ mount
+ grep /dev/mapper
/dev/mapper/root_crypt on / type ext4 (rw,relatime,errors=remount-ro,data=ordered)
/dev/mapper/loop1p2 on /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/rootfs type ext4 (rw,relatime,data=ordered)
/dev/mapper/loop1p1 on /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/bootfs type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
++ mount
++ grep /dev/mapper
++ cut -f 3 -d ' '
++ sort -r
+ local 'LOCS=/home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/rootfs /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/bootfs /'
+ echo 'unmount: Debug: Start of LOCS'
unmount: Debug: Start of LOCS
+ echo '/home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/rootfs /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/bootfs /'
+ sed -e 's/^/"/;s/$/"/'
"/home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/rootfs"
"/home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/bootfs"
"/"
+ echo 'unmount: Debug: End of LOCS'
unmount: Debug: End of LOCS
+ for loc in '$LOCS'
+ umount /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/rootfs
+ for loc in '$LOCS'
+ umount /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs/bootfs
+ for loc in '$LOCS'
+ umount /
umount: /: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)

real    16m50.569s
user    35m33.560s
sys     0m37.064s
```

And the problem is now clear I think.  Brilliant suggestions regarding this would be appreciated--I can dink with it and test patches in the meantime.

The moment I saw...

```
+ unmount /dev/mapper . /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs . . . /dev/mapper . /home/tjcarter/Source/iKarith/pi-gen/work/2016-09-16-Raspbian/export-noobs . . .
```

...I had some idea of the nature of the problem.

iKarith, 2016-09-17:
Okay, I've been trying to wrap my head around this a little bit, and I think I can see some of where the problems lie.  First, let's have a look at unmount_image:

```bash
unmount_image(){
	sync
	sleep 1
	local LOOP_DEV=$(losetup -j ${1} | cut -f1 -d':')
	if [ -n "${LOOP_DEV}" ]; then
		local MOUNTED_DIR=$(mount | grep $(basename ${LOOP_DEV} | head -n 1 | cut -f 3 -d ' '))
		if [ -n "${MOUNTED_DIR}" ]; then
			unmount $(dirname ${MOUNTED_DIR})
		fi
		sleep 1
		kpartx -ds ${LOOP_DEV}
		losetup -d ${LOOP_DEV}
	fi
}
```

First of all, the way this gets used in `export-noobs/prerun.sh` appears to not be the way it was designed to be used.  It's intended to unmount a tree of mounts (in itself no simple task), not really two parallel mounts.  This is evidenced by the attempt to build MOUNTED_DIR, the root of that tree.  So let's see what this actually does here.

Okay, losetup -j on the image filename will return a the device node followed by a : and other things we don't care about, so we cut that to get /dev/loop0.  Works as expected, and if this returns nothing, we do nothing.  Fair enough.

And then we set MOUNTED_DIR and I see our first bug:

```bash
local MOUNTED_DIR=$(mount | grep $(basename ${LOOP_DEV} | head -n 1 | cut -f 3 -d ' '))
```

We are grepping for `basename /dev/loop0 | head -n 1 | cut -f 3 -d ' '`.  I get:

```
/dev/mapper/loop0p1 on /home/tjcarter/Source/iKarith/pi-gen/deploy/bootfs type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
/dev/mapper/loop0p2 on /home/tjcarter/Source/iKarith/pi-gen/deploy/rootfs type ext4 (rw,relatime,data=ordered)
```

I bet it's supposed to look more like this:

```
tjcarter@rem:~/Source/iKarith/pi-gen/deploy$ mount | grep $(basename /dev/loop0) | head -n 1 | cut -f 3 -d ' '
/home/tjcarter/Source/iKarith/pi-gen/deploy/bootfs
```

Which is already not going to do what you want here because you expected this to be a nested mount, and it isn't.  Let me remount that in a nested form and `mount | grep loop0`:

```
/dev/mapper/loop0p2 on /home/tjcarter/Source/iKarith/pi-gen/deploy/rootfs type ext4 (rw,relatime,data=ordered)
/dev/mapper/loop0p1 on /home/tjcarter/Source/iKarith/pi-gen/deploy/rootfs/boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
```

Yes, that's going to produce ...

```
tjcarter@rem:~/Source/iKarith/pi-gen/deploy$ mount | grep $(basename /dev/loop0) | head -n 1 | cut -f 3 -d ' '
/home/tjcarter/Source/iKarith/pi-gen/deploy/rootfs
```

NOW unmount_image should do what it's supposed to.  Of course that won't help export-noobs much without a small patch.  Two are possible.  Here's the code in question that needs to change is:

```bash
mkdir -p ${STAGE_WORK_DIR}/rootfs
mkdir -p ${STAGE_WORK_DIR}/bootfs
mkdir -p ${NOOBS_DIR}

mount $ROOT_DEV ${STAGE_WORK_DIR}/rootfs
mount $BOOT_DEV ${STAGE_WORK_DIR}/bootfs

tar -I pxz -C ${STAGE_WORK_DIR}/bootfs -cpf ${NOOBS_DIR}/boot.tar.xz .
tar -I pxz -C ${STAGE_WORK_DIR}/rootfs -cpf ${NOOBS_DIR}/root.tar.xz .
```

Two options are possible.  Here's the first:

```bash
mkdir -p ${STAGE_WORK_DIR}/rootfs
mkdir -p ${NOOBS_DIR}

mount $ROOT_DEV ${STAGE_WORK_DIR}/rootfs
tar -I pxz -C ${STAGE_WORK_DIR}/bootfs -cpf ${NOOBS_DIR}/root.tar.xz .

mount $BOOT_DEV ${STAGE_WORK_DIR}/rootfs/boot
tar -I pxz -C ${STAGE_WORK_DIR}/rootfs/boot -cpf ${NOOBS_DIR}/boot.tar.xz .
```

Option two looks like this:

```bash
mkdir -p ${STAGE_WORK_DIR}/rootfs
mkdir -p ${NOOBS_DIR}

mount $ROOT_DEV ${STAGE_WORK_DIR}/rootfs
mount $BOOT_DEV ${STAGE_WORK_DIR}/rootfs/boot

tar -I pxz -C ${STAGE_WORK_DIR}/rootfs/boot -cpf ${NOOBS_DIR}/boot.tar.xz .
tar -I pxz -C ${STAGE_WORK_DIR}/bootfs --exclude=./boot -cpf ${NOOBS_DIR}/root.tar.xz .
```

I've chosen the latter as the cleaner of the two.  A build is running now.  If it is successful, I'll commit it which will make my currently open cosmetic PR no longer cosmetic.  If you'd like me to revert the cosmetics and/or apply the first option. please let me know.

iKarith, 2016-09-17:
I just realized something: I made it exclude ./boot, which is going to exclude the directory, as well as its contents, leaving you without a /boot mountpoint.  That won't do at all, I'll fix that before committing.

iKarith, 2016-09-24:
Actually, I have a better solution for this I'm testing now.  Rather than trying to use a loop and sort to try and get a list of things to unmount in reverse order, why not use umount -R?  Never occurred to me to try that.

iKarith, 2016-09-24:
...because I'm an idiot, that's why.  *facepalm*

XECDesign, 2016-09-24:
I originally used -R, but found that it's not available in the version of umount on my main distro (Mint).

Sorry about lack of activity from me here lately. Will address everything after the next release. No major problems with anything I've seen.

I didn't make the sourced scripts executable, give them a shebang or executable permissions since they're sourced by another script, which Is executable. I think I read that as a suggestion in a bash scripting style guide. I am not set on that approach, as long as things are consistent (which they currently aren't).

iKarith, 2016-09-24:
I didn't make them executable or give them a shebang line, just a .sh suffix and the word bash in the first line (plain comment) which some editors like vim use as a context clue that bashisms are not syntax errors.

I won't add -R to this PR if it's not available on your distribution. It didn't really solve my problem completely anyway—the complex scripts I didn't write assume it's safe to just start daemons and mount fuse filesystems. Until I rewrite them I need to shut that down before -R solves anything.

Joseph
Sent via mobile

iKarith, 2016-10-06:
Just wondering if you've had a chance to check out the patch in the above-mentioned PR.  I've been using it now for a little while and it's solved the problem I had.

XECDesign, 2016-10-06:
Had a chance to glance, but not go through it thoroughly and test.

PR #22 (2016-10-06): Fix umount_image
Closes #19.

The first problem is found within unmount_image.  It appears that a ) was misplaced, feeding garbage to the rest of the pipe stream to figure out what to unmount.  Fixing this causes the unmounts to properly succeed, because the umounts all succeed and the commands to request the image be detached appear to complete successfult.  This fixes half the problem, the script completes correctly, and it all looks good.

Then you notice the loop device is still connected.  That's because export-noobs violates the precondition of unmount_image: The image is not associated  [... cut]
closed 2016-10-06 by XECDesign