RPi-Distro/pi-gen #23: [Suggestion] ${i}-run.sh must be .sh?

iKarith, 2016-09-30:
Noticed a few things looking at build.sh and scripts/common:

1. ${i}-run-chroot (no suffix) is a thing that could be used, but never is even where it makes sense.
2. ${i}-run.sh (.sh suffix) is used, and could be any language of course, though bash is obviously assumed.
3. on_chroot everywhere uses sh -e (dash) rather than bash -e.

While contemplating what if anything to do about any of the above, it crossed my mind: What if for some reason you wanted or needed to have a -run script that wasn't written in bash?  How would that work?

Answer: Something like this:

```bash
#! /bin/bash

i=00

for run_script in ./${i}-run{.*,}; do
        if [ -x "$run_script" ]; then
                echo "Begin running ${run_script}"
                $run_script
                echo "End running ${run_script}"
        fi
done
```

Okay, let's try that out:

```
00-run.py: a /usr/bin/env python3 script, ASCII text executable
00-run.sh: Bourne-Again shell script, ASCII text executable
00-run:    a /bin/cat script, ASCII text executable
```

... /bin/cat!?  Sure, why not.

```
Begin running ./00-run.py
./00-run.py has been run!
End running ./00-run.py
Begin running ./00-run.sh
./00-run.sh has been run!
End running ./00-run.sh
Begin running ./00-run
#! /bin/cat

I'm pretty useless, but I could exist and be anything at all...
End running ./00-run
```

At all wanted or cared about, or is this feature creep?  I've got thoughts about how -run-chroot could be useful, if it were a file copied to the rootfs, run there, and then deleted, which would allow for more flexibility than an inline script.  That's particularly useful to me for now since my Raspple II stage 5 presently uses -run to arbitrarily run random scripts from the chroot, though how that's done will change over time.

XECDesign, 2016-10-04:
I think this would fall into the feature creep category until I see a case where this would be useful for raspbian.

iKarith, 2016-10-05:
Fair enough--I'll leave that be for now and close this issue.

I still need multi-arch builds which is also something you don't _yet_ need--but that's at least something you guys are starting to consider a possibility when the Raspberry Pi 5B ships with 8GB RAM and a quad 4GHz arm64 all for $35 and it actually starts to make sense to build arm64 Raspbian.  :)   (ha ha, yeah, we all wish...)

I can think of a couple of ways to do it and pros and cons of each.  I'll open an issue to discuss competing proposals when I've got some.  Not ready to write the code yet, and until I am, ... yeah.  :)

closed 2016-10-05 by iKarith

XECDesign, 2016-10-05:
Multi-arch builds wouldn't hurt and wouldn't necessarily add too much complexity. I guess the thing to decide is whether you would rant to do something like ARCH=arm64 ./build.sh, or have a separate branch with the arm64 config. If it's the first case, then we'd often need slightly different package lists. For example `packages` as the core packages and `packages.$ARCH` which are only installed on the given architecture. I suppose the same approach could be used for patches and scripts too.

iKarith, 2016-10-05:
Oh, definitely need different package list support. After all, my fist other arch to support is i386 Debian. Probably soon followed by amd64. They don't have your custom packages which I'd want to install on a Raspbian based system.

It really comes down to whether I want to copy/maintain two different trees of stage files (which means making pi-gen support that) or if I'd rather have one stage tree with files used only for a build on a specific arch.

Both have pros and cons, but it's late for UTC-0800 and I'll be busy for a couple of days, so I have time to think on it.

Joseph
Sent via mobile

XECDesign, 2016-10-06:
Just remembered lb's approach to this. Their package lists have conditionals like this:
```
#if ARCHITECTURES armel
somepackage
#endif
```
I prefer that to adding even more files to keep track of, although it conflicts a little with the comments.

iKarith, 2016-10-06:
Despite the similarity to cpp, I believe they are not using cpp to process #if, so it's just a matter of us replacing the comment stripper with something more advanced.  I'll look into the conditionals and see how that's done.  I'd figured it'd be unwanted additional complexity to do it that way, which is why I didn't suggest it outright.  But it's a valid third option, and I'll investigate it as well.

It's actually also reasonable to begin discussing Debian's live-build system directly.  We could begin transitioning pi-gen's stage directories to be more and more like lb's configuration to the point that should lb ever gain support for armhf, an image build could be assembled using nothing but Debian's standard tools.  We might need to patch something to support doing things for NOOBS, but again it's reasonable that it be done.  We could probably add the necessary armhf support to the lb tools to push that along.

The major reason not to do it (at least not yet) is the state of live-build in Debian.  If you're familiar with the live-wrapper package and why it was created, the live-build dev's rage quitting Debian and declaring live-build officially dead, and the subsequent reintroduction of (I believe) open-infrastructure-system-build which is just live-build in more corporate industry working group collaboration clothing on a website whose software offerings all claim him alone as their author.

The new really-long-package-name lb conflicts/replaces the old one, and is now being developed without any desire or intent to have it be something related to Debian other than by accident of that's the Linux distribution this working group chose.

Meanwhile live-wrapper emulates lb when it comes to reading lb's configuration, supports any architectures Debian does--just drop in support for the necessary bootloaders and the like, and will replace live-build.  It's still an immature tool, and it was developed if not in secret as was accused, pretty intentionally out of the live-build developer's attention, which is why he accused them of conspiring against him and rage quit and all in the first place.

It was a common saying amongst DPLs back when I was a developer with the project that running Debian is like attempting to herd cats.  I see this remains true.  :)  Pi folks might do well to keep doing our own thing until the dust settles a bit more.