RPi-Distro/pi-gen #27: lightdm.conf patch fails at stage4

whiterocker, 2016-10-25:
Using the 2016-09-23 release, the patch stage4/02-extras/00-patches/0-autologin.diff attempts to set lightdm for autologin-user=pi, but at least on my build, it fails because that has already been set at stage3.
Not sure what modified /etc/lightdm/lightdm.conf at stage3.  The config file as delivered in lightdm_1.18.2-2_armhf.deb has autologin-user commented out as expected.  Nothing obvious in the logs indicate that line being modified by another package install or by an explicit action of stage3.

XECDesign, 2016-10-25:
Thanks, I will look into it. There may currently be a few other issues, which will get resolved by the next release.

XECDesign, 2016-11-01:
Should be fixed by 09083329663cb9354f2ef56f9db2067d544c9701

closed 2016-11-14 by XECDesign