RPi-Distro/pi-gen #28: Slow builds

dsigurds, 2016-11-02:
Building an image takes a very long time.  Are there any suggestions for areas where improvements could be made to speed up the process.

XECDesign, 2016-11-03:
On your end, you could combine stages, skip rebuilds of stages which don't need to be rebuilt and remove exports you don't need. On our end, certain things could be done in parallel and copying could be avoided by using qcow image branches. Most of the time is spent installing packages, and there's not a whole lot that can be done about that, since dpkg wasn't designed to utilise multiple cores.

However, build time is not a priority for me and reducing it properly would result in too much unnecessary complexity, so I don't think that there will be much work done in that area.

dsigurds, 2016-11-03:
thanks for the suggestions.  I've used some of the techniques (particularly the skipping rebuilds of stages) but I like being able to do a full build from scratch when doing a release.  I understand how this probably isn't super critical.  was just checking if there was any low hanging fruit.  Thanks, and thanks for developing and open sourcing this project.

XECDesign, 2016-11-03:
No problem, but 99% of the work was done by asb for spindle. I've just refactored it in a way that suits me better.

closed 2016-11-14 by XECDesign