RPi-Distro/pi-gen #31: Question: What to build on

StaticCube, 2016-11-16:
Hello All,

Not really a bug (unless filled under documentation update request) but my question is when building a raspberry pi image can this be done on any Linux PC or does it need to be build on the pi itself (I know certain dependencies need to be installed)?

To not finish with a question 3 quick things I noticed:
- Tried doing a build which gave me some errors (only to realize I was starting to script with sh build.sh instead of bash build.sh)
- config file contents where unclear to me (figured it out after reading some previous bugs might be nice to add this to the docs :-))
- After checking out I noticed the prerun.sh scripts weren't running after a bit of research noticed the sh files needed to be chmodded to have execute rights else they aren't executed.

Not really good things but would be nice stuff to add to the docs (ow yeah and of course my question ;-))

With kind regards

StaticCube

Landrash, 2016-11-17:
*Not a full answer to your feedback but my use-case.*
I've built this on a Digital Ocean droplet running Debian without any problems. 
I create the droplet, install git ,clone my repository and build the image. When I'm done I destroy the droplet and it costs me a few cents each time.

XECDesign, 2016-11-17:
Just run sudo ./build.sh. The shebang is there for it to run properly.

It should build on Mint running on a desktop or raspbian on a pi. I only use Mint, so  building on a pi may or may not work.

The files which need to be executable should already be executable. You may have a git configuration issue.

If there is something that should be in documentation but isn't, please send a PR.

cashpipeplusplus, 2016-11-30:
I'm building on Ubuntu 14.04.  Not sure if the host system needs to be Debian-based (Debian, Ubuntu, Mint, etc), but it definitely does *not* need to be built on the Pi.

I haven't tried running pi-gen on the Pi itself, but it already takes roughly forever to build an image from scratch w/ 12 cores and 16 GB RAM.  I imagine it would be not be a good use of time to run pi-gen on the Pi.

XECDesign, 2016-11-30:
The number of cores and RAM doesn't make much of a difference. Most of the time goes to dpkg, which only uses a single core and runs through qemu. Whatever you gain from a faster processor, you lose through emulation.

cashpipeplusplus, 2016-11-30:
Ah, I see.  Thanks!

StaticCube, 2016-11-30:
Thanks All, I guess I can mark this issue as resolved :-)

closed 2016-11-30 by StaticCube