RPi-Distro/pi-gen #32: Question: read-only root filesystem

cashpipeplusplus, 2016-11-30:
I'm trying to modify pi-gen to create an image which mounts `/` and `/boot` read-only.  This means, for example, removing `resize2fs_once`.  In a nutshell, I'm trying to take the steps in [this article](https://hallard.me/raspberry-pi-read-only/) and do them at generation time so that the system is read-only from the very first boot.

Has anyone tried this, or am I in new territory?  I'm sure I'll get it working eventually, but I wanted to see if anyone has any experience here before I spend too much time hacking away at it.

XECDesign, 2016-12-06:
Looks like there isn't anybody else doing this. I'll close this since it's not actually an issue.

closed 2016-12-06 by XECDesign

cashpipeplusplus, 2016-12-07:
Fair enough.  For what it's worth: https://github.com/cashpipeplusplus/pi-gen/commit/b3cfaddfec7208f417dbd97b8c6a371d3a94a880#diff-c59b8b8fa77cc3c4ea5682ccf394fe53

cashpipeplusplus, 2016-12-07:
Also, is there a better place for me to ask questions?  A mailing list or forum or something that would be preferable to github's issue tracker?

XECDesign, 2016-12-08:
pi-gen doesn't exactly have an active user base to justify a mailing list or a forum. I am happy to answer questions here, but if I don't have an answer and nobody else chimes in, I will close the issue after a while.

derpston, 2016-12-10:
I'd definitely use this feature if it were available. I'll try to work @cashpipeplusplus' existing work into a PR with an option controlled by the config file. Is that something that you'd consider merging?

cashpipeplusplus, 2016-12-10:
JFYI, if read-only were to be an option, there would be better ways to do it than what I did in my fork.  For example, in my fork, I made it a goal replace stage 3 and leave the rest as-is, to make rebasing easier.  That means I'm undoing certain parts of earlier stages, rather than changing those stages themselves.

XECDesign, 2016-12-10:
The idea is that the stages themselves are separate from build system. The stages provided here are a starting point for people who want to customise the official raspbian image for their needs or to track development and see upcoming changes.

I don't see how a read only option in config fits in with the above. I think it works best as a fork.