RPi-Distro/pi-gen #34: Build failed at stage 4

hj91, 2016-12-04:
ownership of `/home/harshad/pi-gen-dev/work/2016-12-04-harry/stage4/rootfs/home/pi/python_games' retained as harshad:harshad

rsync: - - chown=1000:1000: unknown option
rsync error:syntax or usage error (code 1) at main.c(1453) [client=3.0.9]

What has gone wrong here?

XECDesign, 2016-12-04:
What's the host distro (and version) you're using?

hj91, 2016-12-04:
It is 64 bit Ubuntu 14.10 and x86 platform.

XECDesign, 2016-12-05:
What does 'rsync --version' report?

hj91, 2016-12-05:
It's 3.0.9

--sent from mobile device--

XECDesign, 2016-12-05:
I'm on 3.1.0:

>     - Added the --usermap/--groupmap/--chown options for manipulating file
>      ownership during the copy.
https://download.samba.org/pub/rsync/src/rsync-3.1.0-NEWS

hj91, 2016-12-05:
OK, I will try it.

--sent from mobile device--

hj91, 2016-12-06:
Updated rsync to latest version and managed to build default raspbian lite and raspbian full image. 

Now to customize its artwork and other related packages that are bundled by default. 

Is it possible to modify those images they show during installation!? For eg - I do not want to pack wolfram in my image. 

Please suggest a workaround for changing logo and icons. Want to include opensource and free stuff only.

XECDesign, 2016-12-06:
To remove a package, just remove it from the stage's package list. If you'd like to modify a package, you're mostly on your own there. I would put custom packages into a separate repo and override the ones in raspberrypi.org.

You can use the icons under the BSD 3-clause license.

closed 2016-12-06 by hj91