RPi-Distro/pi-gen #39: apt-get failures

uSpike, 2017-01-04:
This may be an issue with the `archive.raspberrypi.org` server, but I'm inconsistently seeing build errors:

```
Now at patch 02-persistant-net.diff
[22:43:40] End /work/stage1/02-net-tweaks/00-patches
[22:43:40] Begin /work/stage1/02-net-tweaks/00-run.sh
Adding 'local diversion of /lib/udev/rules.d/75-persistent-net-generator.rules to /lib/udev/rules.d/75-persistent-net-generator.rules.distrib'
[22:43:40] End /work/stage1/02-net-tweaks/00-run.sh
[22:43:40] End /work/stage1/02-net-tweaks
[22:43:40] Begin /work/stage1/03-install-packages
[22:43:40] Begin /work/stage1/03-install-packages/00-packages
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  alsa-utils device-tree-compiler libasound2 libasound2-data libparted2 libsamplerate0 lua5.1 parted triggerhappy
Suggested packages:
  libasound2-plugins libparted-dev libparted-i18n parted-doc
The following NEW packages will be installed:
  alsa-utils device-tree-compiler libasound2 libasound2-data libparted2 libraspberrypi-bin libraspberrypi0 libsamplerate0 lua5.1 parted raspi-config triggerhappy
0 upgraded, 12 newly installed, 0 to remove and 0 not upgraded.
Need to get 4384 kB of archives.
After this operation, 11.4 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian/ jessie/main libasound2-data all 1.0.28-1+rpi1 [65.1 kB]
Get:2 http://archive.raspberrypi.org/debian/ jessie/main libasound2 armhf 1.0.28-1+rpi1 [321 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main libparted2 armhf 3.2-7 [259 kB]
Get:4 http://archive.raspberrypi.org/debian/ jessie/main device-tree-compiler armhf 1.4.1-1+rpi1 [351 kB]
Get:5 http://mirrordirector.raspbian.org/raspbian/ jessie/main libsamplerate0 armhf 0.1.8-8 [939 kB]  
Get:6 http://mirrordirector.raspbian.org/raspbian/ jessie/main alsa-utils armhf 1.0.28-1 [960 kB]
Err http://archive.raspberrypi.org/debian/ jessie/main libraspberrypi0 armhf 1.20161215-1
  404  Not Found [IP: 93.93.130.104 80]
Get:7 http://mirrordirector.raspbian.org/raspbian/ jessie/main lua5.1 armhf 5.1.5-7.1 [75.1 kB]
Get:8 http://mirrordirector.raspbian.org/raspbian/ jessie/main parted armhf 3.2-7 [191 kB]
Err http://archive.raspberrypi.org/debian/ jessie/main libraspberrypi-bin armhf 1.20161215-1
  404  Not Found [IP: 93.93.130.104 80]
Get:9 http://mirrordirector.raspbian.org/raspbian/ jessie/main triggerhappy armhf 0.3.4-2 [31.2 kB]
Get:10 http://archive.raspberrypi.org/debian/ jessie/main raspi-config all 20161207 [17.7 kB]   
Fetched 3211 kB in 2s (1154 kB/s)                          
E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20161215-1_armhf.deb  404  Not Found [IP: 93.93.130.104 80]

E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20161215-1_armhf.deb  404  Not Found [IP: 93.93.130.104 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

uSpike, 2017-01-04:
It seems to work better if I add a call to `apt-get update` to the tune of:

```
        if [ -f ${i}-packages-nr ]; then
            log "Begin ${SUB_STAGE_DIR}/${i}-packages-nr"
            PACKAGES="$(sed -f "${SCRIPT_DIR}/remove-comments.sed" < ${i}-packages-nr)"
            if [ -n "$PACKAGES" ]; then
                on_chroot << EOF
apt-get update
apt-get install --no-install-recommends -y $PACKAGES
EOF
            fi
            log "End ${SUB_STAGE_DIR}/${i}-packages-nr"
        fi
        if [ -f ${i}-packages ]; then
            log "Begin ${SUB_STAGE_DIR}/${i}-packages"
            PACKAGES="$(sed -f "${SCRIPT_DIR}/remove-comments.sed" < ${i}-packages)"
            if [ -n "$PACKAGES" ]; then
                on_chroot << EOF
apt-get update
apt-get install -y $PACKAGES
EOF
```

XECDesign, 2017-01-06:
There was a problem with one of the load balances being out of sync, which should be fixed now. Has it resolved this issue?

uSpike, 2017-01-11:
Things are good now, thanks!

closed 2017-01-11 by uSpike