RPi-Distro/pi-gen #45: Remove dependency on kernel support for binfmt

sam3d, 2017-02-07:
In order to build for docker, the underlying kernel has to have support for `binfmt`, which is required by `qemu-user-static`. On Docker for mac, this works fantastically because the kernel already has the required support. However, when attempting to run on a fresh Debian VPS many kernels don't have support.

I'm not sure if there's an easy solve. Many have suggested over the internet to use [a version of qemu with the `QEMU_EXECVE` set so compiling for ARM doesn't escape out and cause the build to fail](https://github.com/resin-io/qemu). A [special docker image](https://github.com/show0k/docker-miniconda-armv7) has been created that uses this prebuilt special version of qemu but I'm not sure how much use this is.

Attempting to install qemu-user-static produces the following output:

```
Setting up qemu-user-static (1:2.1+dfsg-12+deb8u6) ...
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
...
```

And some people on `circleci` have documented [similar behaviour](https://discuss.circleci.com/t/add-binfmt-misc-support-for-cross-compilation/2264/3) as a failed result of cross-compilation for ARM architectures.

While this isn't an issue inherent to `pi-gen`, it would be good to remove the dependency on underlying kernel support that isn't always present.

sam3d, 2017-02-07:
I think I've found a temporary workaround to this issue.

On the host machine (outside the docker container), you need to simply install `binfmt-support` there. It cannot be done from within the container (no matter the privileged mode). Not quite sure how to get around this issue though, as it appears to be a requirement.

XECDesign, 2017-02-08:
Not sure that this is an issue rather than just a limitation. Pi-gen only really supports whatever OS I happen to be running at the time and anything outside of that is too much headache, due to subtle differences like this.

sam3d, 2017-02-08:
That's entirely fair; at least now _this_ is documented I'd say it's safe to close it. Touch wood that this is one of the more obscure OS/kernel - dependent issues now that building with docker is in place.

closed 2017-02-08 by sam3d