RPi-Distro/pi-gen #47: First boot configuration file/script

sam3d, 2017-02-08:
This'd be a relatively major feature, and I know it's been suggested before but I'm unsure if there's an intentional reason it hasn't been implemented (and if so feel free to ignore the rest of this proposal).

## Description of the problem
There are many reasons this would be beneficial, but the most commonly cited issue is **configuring a Raspberry Pi through just the use of wifi**. As it stands right now, you'd have to do one of the following:

- Compile your own image with the wifi configuration baked in
- Use a third party utility that relies on custom images such as `PiBakery`
- Connect ethernet on initial boot to add appropriate wifi configuration
- Use an OS with write support for `ext4` (neither Mac or Windows can do this natively)

All of the above solutions have frustrating aspects to them; a native solution that requires no third party utilities or faffing around with ethernet cables would be ideal.

## Feature proposal

In essence, a configuration file/script placed in the `boot` partition of the SD card. It's `FAT` which means that almost every operating system in existence can read _and write_ to it natively.

Much like with the empty `ssh` file, this would only take effect on first boot (thus hopefully eliminating many of the security concerns with having such a configuration file).

### File specifics
This could theoretically be implemented in two ways:

1. A `config.json`, `config.txt` or `config.yml` with structured config data
2. A `config.sh` of some kind that simply executes on first boot

It could be either/or, as there are advantages and disadvantages to both. A structured config file would be easier to setup and have fewer security concerns (with the limited number of properties that could be modified), but with the greater overhead of having to parse the file contents.

Likewise a `config.sh` would be slightly harder to configure but offer the user much greater customisability over their Pi. The security concerns would be slightly more-so but if the user takes care to make sure it's freshly written image (with validated checksums) it's unlikely it could be used maliciously. It'd be fast though.

Perhaps the option could be allowed for both, so that a user can take whichever approach they are most comfortable with.

### Implementation
Placing a file that executes the script (or parses the structured config data) in `/etc/rc.local` or `/etc/init.d/...` and then deletes itself so that it is guaranteed to only run once would probably be the best way to go.

-----

I personally think this would be a fantastic feature to have natively, and would eliminate a lot of headache with setting up a new Raspberry Pi. I would be more than happy to do all of the scripting, implementation and documentation for it myself. Just wanted to run it by you and see what your thoughts were before attempting 😄

XECDesign, 2017-02-08:
https://github.com/RPi-Distro/raspberrypi-net-mods/blob/master/01-wpa-config-copy ?

sam3d, 2017-02-08:
Ah. There is a chance I may have overlooked that. My bad.

sam3d, 2017-02-08:
Okay so moving aside the WiFi case of my argument entirely (which is admittedly, like, all of it). Is this a feature that would be worth fleshing out?

I still can't help but feel as though things such as automatic/unattended configuration of the hostname, the main username, their password, ssh authorized_keys, etc. would be worth the creation of such a feature.

XECDesign, 2017-02-08:
Yes, but I think it would need to be properly fleshed out before it's implemented. I am thinking something like pi-gen's build.sh that runs a stage from /boot/something on boot, but that's something I'd need to discuss with others at pi towers. There are reasons to avoid doing it that way. A simple shell script would be much simpler for most people.

It isn't a priority because you can easily modify an existing image, run a script over ssh to do what you need or customise pi-gen to create exactly the image you want. Pi-gen stages are probably a little complicated for the sort of person that doesn't find any of the above options acceptable. In that case something like PiBakery is probably perfect anyway.

There are some security concerns too. You could argue that if you have access to /boot, you probably already have access to everything. In this case a .txt config file with limited options is probably a little better.

I'd hold off on implementing anything at all, until the best approach is agreed on.

closed 2017-03-06 by XECDesign

procount, 2017-03-06:
See https://github.com/procount/noobsconfig - it does the basics. Meltwaters pikitchen adds the run only once on 1st biot functionality and more. Would be better with a pibakery style interface though...