RPi-Distro/pi-gen #49: Best practices to customise it?

fsamir, 2017-02-21:
Hi all,
the project doesn't seem to have discussion group, so I will ask here.

1. Given I am working on my own step and I sometimes make mistakes, is there a way to avoid running all previous steps and speed up the development process?

2. If I was to use a devops tool like Ansible or Chef, what would be the best approach?

Thanks,
Franklin

sam3d, 2017-02-21:
If you're looking to avoid running previous steps, you could just throw a `SKIP` file in the root of any steps you'd like to not run. I don't usually wanna run steps 3 & 4, so I'd use `touch step{3,4}/SKIP`.

As for the devops tool, well that depends. What sort of thing are you trying to automate? The installation of Ansible or Chef onto the ISO? Using Ansible and Chef to configure a system to run the `pi-gen` utility?

fsamir, 2017-02-21:
Hi @sam3d 

1. Thanks fot the advice. I am using SKIP for the steps I never wanna run and I've commented out the check on `build-docker.sh:46`, so I can run with CONTINUE=3|4, when I am developing a new step.

2. I've got a set of Fabric scripts, which I run over SSH to customise an running Raspbian instance. I would like to reuse it, in one of me steps, just so I don't have to convert all of them to bash scripts.
I guess this is a common scenario for others willing to migrate from SSH based customisation, given that building it from scratch has many advantages.

XECDesign, 2017-02-22:
I don't know anything about fabric or how it could integrate with pi-gen. What does the Fabric script do?

adampie, 2017-02-22:
Hi @fsamir,

I've been doing stuff regarding pi-gen and Ansible as a project [here](https://github.com/adampie/pi-gen), I'm still playing around with it but it might give you some ideas.

fsamir, 2017-02-23:
@XECDesign fabric is a low level devops tool, basically ssh on steroids. Yes, the link you suggested is how I am approaching it. That alone doesn't work, because fabric still tries to ssh to localhost, but it is a good start.
Anyway, thanks for the tips. I got a better picture of how the scripts work and the answer to my question is one of these: 
1. Run a devops tool of choice as part of one of the steps (if you are lucky and the tool works). 
2. Or, convert the devops tool scripts to bash scripts.
3.  Or, do a rewrite like @adampie suggested.

#1 is preferable, because I could still use the same scripts on a running instance of Raspbian, over ssh.
Thanks

XECDesign, 2017-02-23:
Does the script do so much that it wouldn't be trivial to port over? I found that the common tasks are usually 'install/replace a file', 'patch a file', 'install a package' and 'run this script', so pi-gen is structured in a way that aims to make it all fairly simple.

fsamir, 2017-02-25:
@XECDesign migrating the scripts wouldn't be too difficult, in my case. However, testing has proven to be a very time consuming task.
By the way, do you know if it is possible to run the image with an emulator (such as qemu), in order to test it without burning to the sdcard?

Thanks

XECDesign, 2017-02-25:
Possible, but a pain in the neck. 
http://web.archive.org/web/20150512213356/http://xecdesign.com/qemu-emulating-raspberry-pi-the-easy-way/
I've abandoned that blog since maintaining it was too much hassle. I would recommend using netboot and nfs for testing.

closed 2017-03-06 by XECDesign