RPi-Distro/pi-gen #50: Building forward from Stage 2

rschulman, 2017-02-26:
I'm looking to go forward from a lite system toward a server-type setup (so, not a graphical direction). Is the right way to do this in pi-gen to just remove the existing Stage 3/4 and build a new Stage 3 that installs what I need with the config I'm looking for, or am I thinking about this wrong?

XECDesign, 2017-02-26:
The approach you suggested is exactly right.

closed 2017-02-26 by rschulman

RichardBronosky, 2017-10-04:
If you create a custom `stage3` and need to do many development cycles of build-test-edit-repeat how can you rebuild only `stage3`?

XECDesign, 2017-10-04:
Drop SKIP files in the previous stages and run sudo CLEAN=1 ./build.sh?

RichardBronosky, 2017-10-04:
This just changed my development feedback loop from 40+ minutes to 2 minutes. Absolutely amazing! Thank you.

RichardBronosky, 2017-10-04:
I was bumping up against bug https://github.com/RPi-Distro/pi-gen/issues/119 and it was taking me 40 minutes to run each test build. The error was in the export, so I moved that to its own stage and skipped all the others. Because @XECDesign helped me, I was able to identify the bug and fix it for others. This is how F/OSS communities work!

jcwren, 2017-10-09:
I'm not having any success with this. I've got my changes in stage3, and it builds perfectly. But if I drop SKIPs in stage0, stage1, and stage2, I get an error of ```Previous stage rootfs not found```.

I grepped all the scripts in the pi-gen distribution, and I can't find anything with the string ```Previous stage``` in it. So what its looking for, and what script is producing this error message?

Here's a log:
```
[Much unpacking and setting up text removed]
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
 ---> 9b5c76ef98ae
Removing intermediate container 96d0e03441c1
Step 4/5 : COPY . /pi-gen/
 ---> 4f953b642c31
Step 5/5 : VOLUME /pi-gen/work /pi-gen/deploy
 ---> Running in 3cb413081c60
 ---> 1954d30e6a3b
Removing intermediate container 3cb413081c60
Successfully built 1954d30e6a3b
Successfully tagged pi-gen:latest
update-binfmts: warning: qemu-aarch64 already enabled in kernel.
[23:52:24] Begin /pi-gen
[23:52:24] Begin /pi-gen/stage0
[23:52:24] End /pi-gen/stage0
[23:52:24] Begin /pi-gen/stage1
[23:52:24] End /pi-gen/stage1
[23:52:24] Begin /pi-gen/stage2
[23:52:24] End /pi-gen/stage2
[23:52:24] Begin /pi-gen/stage3
[23:52:24] Begin /pi-gen/stage3/prerun.sh
Previous stage rootfs not found
```

XECDesign, 2017-10-09:
Can't really comment on docker builds, but the message means that it doesn't find work/<your build>/stage2/rootfs

RichardBronosky, 2017-10-11:
I'll add that the builds are dated. So, if you did a build yesterday and want to extend it today, you have to rename the build folder.

jcwren, 2017-10-11:
> Can't really comment on docker builds, but the message means that it doesn't find work/your_build/stage2/rootfs
This would be located somewhere in the docker container? 

I have a pretty good understanding of the build scripts, and have heavily modified it for my build, but I still don't have my head completely wrapped around the docker environment, the lifetime of containers, images, and how pigen_work and the pi-gen image are related.

cfstras, 2017-10-13:
Here's a rundown:
`pi-gen` (the image) is the docker image that contains the linux environment to build, along with the build scripts themselves.
`pigen_work` is the container (built from the image) that has build volumes associated with it (mounted at `/pi-gen/work`). These contain the actual build work. They're deleted when `pigen_work` gets deleted.
`pigen_work_cont` is a container that gets the same work volumes mounted from `pigen_work` when you do a build with `CONTINUE=1`. The difference is that it can contain different (newer) build scripts than the original `pigen_work` container.

The fact that build folders are dated breaks this a bit, since the `work` folder then can't be continued the next day. Maybe we should consider removing that date when doing docker builds.

filips123, 2019-01-02:
@XECDesign

I am also going to build my system with custom packages and I will replace stage 3 and later with my own.

How can I then keep updates of lower stages?
Should I fork this repository and overwrite lower stages in my fork with changes in this repository? Is it possible to do this better?

cfstras, 2019-01-02:
you can fork the repo and tgen continouosly merge in the master here. just use „ours“ for everything in stage3  - or modify the scripts to call it „stage30“ to avoid conflicts.