RPi-Distro/pi-gen #53: Error -->  sudo: unable to resolve host hassbian

JonMurphy, 2017-03-06:
The hostname file is correct but the hosts file should be updated to match.  The last line in `hosts` should be `127.0.1.1	hassbian` and that will eliminate the `sudo: unable to resolve host hassbian` error.

Thank you!
Jon

```
pi@hassbian:~ $ sudo systemctl restart home-assistant@homeassistant.service
sudo: unable to resolve host hassbian

pi@hassbian:~ $ cat /etc/hostname
hassbian

pi@hassbian:~ $ cat /etc/hosts
127.0.0.1	localhost
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters

127.0.1.1	raspberry

pi@hassbian:~ $ sudo nano /etc/hosts

pi@ha:~ $ cat /etc/hosts
127.0.0.1	localhost
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters

127.0.1.1	hassbian

pi@hassbian:~ $ 
```

XECDesign, 2017-03-06:
When and how was the hostname changed and how does it relate to pi-gen?

m-roberts, 2017-03-06:
You need to modify `stage1/02-net-tweaks/00-patches/01-hosts.diff` as well. This is not a pi-gen issue.

XECDesign, 2017-03-06:
Thanks, @m-roberts

closed 2017-03-06 by XECDesign

JonMurphy, 2017-03-06:
sorry - I followed too many links and got lost along the way!  Jon