RPi-Distro/pi-gen #58: Including `certbot` from jessie-backports fails authentication

rschulman, 2017-03-27:
I'm trying to include `certbot`, the Let's Encrypt certificate client, in stage3. I have included the jessie-backports repository in `sources.list`, but when the script tries to install certbot, the following error occurs:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  dialog python-acme python-certbot python-cffi-backend python-chardet python-configargparse python-configobj python-cryptography python-dialog python-dnspython python-enum34
  python-funcsigs python-idna python-ipaddress python-mock python-ndg-httpsclient python-openssl python-parsedatetime python-pbr python-pkg-resources python-psutil python-pyasn1
  python-pyicu python-requests python-rfc3339 python-setuptools python-six python-tz python-urllib3 python-zope.component python-zope.event python-zope.interface
Suggested packages:
  python-certbot-apache python-certbot-doc python-acme-doc python-configobj-doc python-cryptography-doc python-cryptography-vectors python-enum34-doc python-funcsigs-doc
  python-mock-doc python-openssl-doc python-openssl-dbg python-psutil-doc doc-base python-socks python-setuptools-doc python-ntlm
Recommended packages:
  letsencrypt
The following NEW packages will be installed:
  certbot dialog python-acme python-certbot python-cffi-backend python-chardet python-configargparse python-configobj python-cryptography python-dialog python-dnspython python-enum34
  python-funcsigs python-idna python-ipaddress python-mock python-ndg-httpsclient python-openssl python-parsedatetime python-pbr python-pkg-resources python-psutil python-pyasn1
  python-pyicu python-requests python-rfc3339 python-setuptools python-six python-tz python-urllib3 python-zope.component python-zope.event python-zope.interface
0 upgraded, 33 newly installed, 0 to remove and 46 not upgraded.
Need to get 2313 kB of archives.
After this operation, 11.1 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  python-cffi-backend python-pkg-resources python-configargparse python-six python-idna python-ipaddress python-pyasn1 python-setuptools python-cryptography python-dialog
  python-dnspython python-funcsigs python-openssl python-ndg-httpsclient python-pbr python-psutil python-urllib3 python-requests python-rfc3339 python-mock python-acme python-certbot
  certbot
E: There are problems and -y was used without --force-yes
```

I have tried adding the main Debian Jessie signing key to `apt-key` and running `apt-key update` but neither has worked so far.

Happy for any thoughts!

XECDesign, 2017-03-28:
Have you run apt-get update after adding the key?

rschulman, 2017-03-28:
Yes. Relevant portion:
```
on_chroot apt-key add - < files/raspberrypi.gpg.key
on_chroot apt-key add - < files/debian.gpg.key
on_chroot << EOF
apt-key update
apt-get update
apt-get dist-upgrade -y
```

XECDesign, 2017-03-28:
Afraid I can't help here. When adding the raspberrypi.org key, it all just worked.

XECDesign, 2017-05-10:
@rschulman If you are still experiencing this problem, where did files/debian.gpg.key come from?

rschulman, 2017-05-10:
This was one of those really annoying bugs that just stopped happening one day without me touching anything I would think would affect it. I'll close this but reopen if it comes back for whatever reason.

closed 2017-05-10 by rschulman

XECDesign, 2017-05-10:
Sounds good, thanks.