RPi-Distro/pi-gen #61: Build fails at export-image/04-finalise/01-run.sh

malaire, 2017-04-16:
Building with current git version fails:

```
$ cd pi-gen-dev
$ echo "IMG_NAME='MyRaspbian'" > config
$ ./build-docker.sh
...
[10:25:03] Begin /pi-gen/export-image/04-finalise
[10:25:03] Begin /pi-gen/export-image/04-finalise/01-run.sh
Stopping fake hwclock: saving system time.
Mode:     real
Files:    2257
Linked:   344 files
Compared: 0 xattrs
Compared: 590 files
Saved:    9.44 MiB
Duration: 0.11 seconds
fatal: Not a git repository (or any of the parent directories): .git
Done! Your image(s) should be in deploy/
```

And there is no image in deploy.

XECDesign, 2017-04-16:
I don't use the docker builds myself, so I am not sure I can help with that one. I am guessing the .git directory is missing from the container, so it can't tell what to put in /etc/rpi-issue

malaire, 2017-04-16:
So is it the `update_issue $(basename ${EXPORT_DIR})` command which is failing? Where does that `update_issue` come from?

XECDesign, 2017-04-16:
https://github.com/RPi-Distro/pi-gen/blob/dev/scripts/common#L92

malaire, 2017-04-16:
ok, I found my mistake. I was using downloaded copy of repository, which doesn't have .git directory. Using `git clone` works:
```
$ git clone https://github.com/RPi-Distro/pi-gen.git
$ cd pi-gen
$ echo "IMG_NAME='MyRaspbian'" > config
$ time ./build-docker.sh
```

closed 2017-04-16 by malaire