RPi-Distro/pi-gen #64: No recovery after failure

redjaguar, 2017-04-20:
There is a horrible issue when the script fails from network issues during any stage of the apt-get processes.

When attempting to resume, the error is "WORK_DIR/?/apt' is not a directory or an apt error that cannot seem to be recovered.

Most of the time resuming is not possible, so a removal of the work directory is required.. this causes a cycle of issues since the original failure was a network issue (in this case DNS randomly failing) and the network issue may or may not occur again. I have also seen an error "cannot verify package" from apt.. which seems that it should be re downloaded instead of reporting an error..

Re downloading the packages *every* time the script is run with a new image name, or new date, or changing something else where the work directory has to be removed is very time consuming when trying to debug and build a deployable image that can be put in an embedded environment with no user intervention.

Creating an apt-cache directory for the downloaded packages in work/.cache or something and setting apt to use that instead of the temporary root fs being created is probably a good way around re downloading the same packages unnecessarily so the only packages being downloaded are the ones that failed before - even after cleaning the working image directory.

XECDesign, 2017-04-20:
Originally, there was a caching mechanism, but it has been removed in favour of  'APT_PROXY' with 'apt-cacher-ng'.

The builds were resumable, unless `CLEAN=1` is set, but that was broken a while back. For now you can 'SKIP' the finished stages and re-run from the one that failed.

redjaguar, 2017-04-20:
It still requires manual removal of the directory for the stage that failed. I saw an APT_PROXY setting in a config file but was not aware of 'apt-cacher-ng'.. Might be a good idea to add that into the readme.

With that in mind, it makes it easier to resume, but maybe the script should remove the directory for the stage that failed, and make a note of which stages were successful, so re running just resumes at the next stage not finished.. unless a 'clean' is indicated then restart the entire thing. That way no manual files have to be created and remembered to be removed.

The failure also creates a situation where everything is still mounted, so it has to be unmounted before anything can be removed. 

If I added any of this in locally, would it be useful to submit back?

If not, or at lest for now a note in the readme of what needs to be done if a stage fails would be helpful.

XECDesign, 2017-04-20:
>
> It still requires manual removal of the directory for the stage that 
> failed. I saw an APT_PROXY setting in a config file but was not aware 
> of 'apt-cacher-ng'.. Might be a good idea to add that into the readme.
I build with CLEAN=1 and SKIP successful stages, which overcomes some of 
these problems, but it's far from ideal. There is already a note about 
apt-cacher-ng, but it could help to expand upon.
> With that in mind, it makes it easier to resume, but maybe the script
> should remove the directory for the stage that failed, and make a note
> of which stages were successful, so re running just [...]
The original idea was that you could continue after failing without 
having to remove the directory. For that to work, run.sh scripts must be 
written in a way that does no harm if they're run multiple times. Quilt 
should keep track of the .pc directories (this was broken when 
sub-stages were added and needs to be fixed).

To remove the manual fiddling with files, the above needs to be fixed 
and verified, locks needs to be added, markers to show which stages 
completed and modified files need to be tracked. I've been thinking 
about taking a weekend or two to re-write build.sh in python with some 
of these considerations in mind.
>
> The failure also creates a situation where everything is still 
> mounted, so it has to be unmounted before anything can be removed.
Yes, although if I recall correctly, build.sh should umount everything 
the next time you run it. It would be good to do the right thing in the 
first place and umount everything after failure as well.
>
> If I added any of this in locally, would it be useful to submit back?
Very much so, as long as it doesn't introduce too much complexity.

redjaguar, 2017-04-20:
I apologize.. I did not see the APT_PROXY note .. 

CLEAN=1 doesn't seem to do anything.. it's in the build script, but what exactly is it supposed to do?

XECDesign, 2017-04-20:
Before starting a stage, it clears out the existing files.

redjaguar, 2017-04-20:
Ahh.. okay.. it seems CLEAN=1 is set before export stage in build.sh.. so would need to be set in config to clean out the stage directories.

CLEAN should probably be added to the readme.. it would have saved me from manually cleaning things up.

matthijskooijman, 2019-12-11:
A related problem is that sometimes /proc, /sys, etc. might remain mounted, preventing removal of the work directory without manually unmounting things.

redjaguar, 2020-02-10:
> A related problem is that sometimes /proc, /sys, etc. might remain mounted, preventing removal of the work directory without manually unmounting things.

I had noticed that as well but never thought to mention it. It has been a while since I have had to build a custom Pi image and was wondering if the script issue was still an issue. Since you posted about things still being mounted I take it the issue still persists.