RPi-Distro/pi-gen #65: Changing default language gives errors

redjaguar, 2017-04-20:
Perhaps the locale package should be installed eralier in the process?

Set the locale to en_US.UTF-8 UTF-8

A number of errors appear for several packages before seeing "generating locale"

/usr/bin/locale: Cannot set LC_CTYPE to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_MESSAGES to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_ALL to default locale: No such file or directory

malaire, 2017-04-22:
I get those messages during build even if I don't change any locale settings. Not sure if this is expected?

XECDesign, 2017-04-24:
That's expected and I've never seen any negative effects from it. There is no harm in installing it earlier to avoid the spam though.

commit 3c0b3640 (2017-06-27): stage0: Install locales

fixes #65
closed 2017-06-27 by XECDesign