RPi-Distro/pi-gen #72: should not use UTC time as build date

g-e-r, 2017-05-24:
This took me some time to find out.
 My timezone - PDT - is UTC-7
 So if I start a build and it fails after 5pm, the date will change...
 That happens very often in the US West coast.
 there should be a better strategy for this.

thomas-phillips-nz, 2017-06-01:
In build.sh, line 131, it assigns the `IMG_DATE` variable to the current UTC date _if the IMG_DATE variable is not set_. You should be able to invoke the script using IMG_DATE=xxx ./build.sh if you want to set your own date.

That said, I don't like the fact that the date is UTC either. I propose we change it to always use local system time. It can always be overridden using the IMG_DATE variable.

XECDesign, 2017-06-27:
Fixed by https://github.com/RPi-Distro/pi-gen/pull/77

closed 2017-06-27 by XECDesign