RPi-Distro/pi-gen #73: Please do not default to using Google nameservers

geromueller, 2017-05-30:
See discussion here: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=761658

blacktigersoftware, 2017-05-30:
very interesting read.  

I agree - but for different reasons.  

Having a default nameserver means that misconfigurations may go unnoticed. example:  "I can't see the machine right next to me, but I can get web-pages, so what's wrong????"   -- vs -- "I can't see the machine next to me and I can't resolve web pages, so what's wrong????"  Debian maintainer was put off with the 'privacy' and 'google' aspect.  I suspect a properly configured machine would never have to resort to the default.  And if it did, you would never know that is was not configured correctly and instead chase some other problem.

Landrash, 2017-05-31:
It's a bad plan forking more things than necessary for Raspbian.
The pi-gen code is forkable and you can just change what you prefer when you make your own images.

XECDesign, 2017-06-27:
Closing due to the same objections as those raised in the linked bug report. We may revisit this later.

closed 2017-06-27 by XECDesign