RPi-Distro/pi-gen #74: -bash: startx: command not found

NorbarLP, 2017-06-01:
After using pi-gen and build.sh to construct (what I am to believe) is an image of a desktop Raspbian system and subsequently copying this image onto a 16GB SD card, I am unable to access the GUI on my Raspberry Pi. Attempting to run startx returns the error in the title, whilst attempting to start the GUI on boot via _raspi-config_ boots the pi into the CLI without providing any error messages/feedback.

I am relatively new to Rapsbian, and as such, I have two queries relating to this:

1) Am I missing something obvious in the Raspberry Pi installation process? After loading the image onto an SD card, I have been placing it in to my Raspberry Pi and powering it up.

2) How do I know which Raspbian Stage I have built to? In the "_Stage Anatomy_" section of readme.md, it is stated that the build of Raspbian is divided up into several stages from 0 to 4 inclusive, where stages 3 and 4 refer to the full Raspbian desktop systems as opposed to the Raspbian-Lite system. However, pi-gen contains stages 0-5 inclusive and I was unable to determine what Stage 5 relates to. I have constructed images up to (and including) Stage 3 and Stage 4 of pi-gen, and I experience the same issue detailed above with both images.

Thank you for all the great work!

ZanderBrown, 2017-06-01:
Can't say what Stage 5 does but if only compleating to Stage 4 is causing problems why not try going through to 5?

NorbarLP, 2017-06-01:
Completing either Stage 3 or Stage 4 causes the same problem. Ideally, I wish to avoid going through to 5 as I am trying to produce a slimmer Raspbian system with a GUI than the complete Raspbian system. I thought it would be more efficient to start from the desktop system, rather than to start from the complete Raspbian system and manually remove hundreds of packages.

thomas-phillips-nz, 2017-06-01:
Are you sure you're generating the images properly. Two images will be created, one for stage2 and the other for stage4. You are loading the image created at stage4, right?

NorbarLP, 2017-06-02:
Apologies for what may be a foolish question, but how do you determine which image is created at which stage?

thomas-phillips-nz, 2017-06-02:
The `EXPORT_IMAGE` file in `stage2/` and `stage4/` specifies the suffix for the generated image.
For example, `stage2/EXPORT_IMAGE` specifies `"-lite"`, so the resulting image file ending in `-lite` in the `deploy/` directory will be the image generated by Stage 2. `stage4/EXPORT_IMAGE` specifies `"-4GB"`, so the image file ending in `-4GB` will be the image file created by Stage 4 (or generated by Stage 5 if you didn't skip Stage 5)

NorbarLP, 2017-06-02:
I apologise, I was using the incorrect image. Thank you for your time and support.

closed 2017-06-02 by NorbarLP