RPi-Distro/pi-gen #78: ioctl failures

titixbrest, 2017-06-05:
When doing successive build, I get ioctl errors using docker or build.sh script.
I tried using virtualbox or a native debian distro

I think that the chroot created is not properly unmounted.

thomas-phillips-nz, 2017-06-05:
Might need to see a full log, can you attach one please? (as a file attachment)
Did it work the first time you ran the script? Or did you kill the script in the middle of running?

titixbrest, 2017-06-05:
I can't do it right now, but I'll try to do this tomorrow on a brand new VM.

titixbrest, 2017-06-06:
[on the VM.log.txt](https://github.com/RPi-Distro/pi-gen/files/1053707/on.the.VM.log.txt)

This is the latest log file. It succeeded one time, and then it failed with the errors on ioctl.

I did not noticed that there are errors for binfmt. Is there a link ?

titixbrest, 2017-06-06:
I did not find out why the problem occured, but I think your code is not guilty. 
I tried on a new virtual machine ans it is now OK.

closed 2017-06-06 by titixbrest

thomas-phillips-nz, 2017-06-07:
I did see some permissions problems with `mount` in your log. Might be related to the VM.
But if it's working now, that's all good