RPi-Distro/pi-gen #80: Python_Openzwave autoinstall fails (HASSBian 1.23)

Vexamus, 2017-06-07:
Forgive my pull req...  Should have been an issue.

Fresh flash of hassbian 1.23
added `zwave:` to configuration.yaml
Receive the following error in log:
```
Jun 07 12:35:49 hassbian hass[738]: 2017-06-07 12:35:49 INFO (Thread-4) [homeassistant.util.package] Attempting install of python_openzwave==0.4.0.31
Jun 07 12:38:19 hassbian hass[738]: 2017-06-07 12:38:19 ERROR (Thread-4) [homeassistant.util.package] Unable to install package python_openzwave==0.4.0.31: Command "/srv/homeassistant/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-92041o8k/python-openzwave/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-uses1i9y-record/install-record.txt --single-version-externally-managed --compile --install-headers /srv/homeassistant/include/site/python3.4/python-openzwave --home=/tmp/tmpyxonxd5x" failed with error code 1 in /tmp/pip-build-92041o8k/python-openzwave/
Jun 07 12:38:19 hassbian hass[738]: 2017-06-07 12:38:19 ERROR (MainThread) [homeassistant.setup] Not initializing zwave because could not install dependency python_openzwave==0.4.0.31
Jun 07 12:38:19 hassbian hass[738]: 2017-06-07 12:38:19 ERROR (MainThread) [homeassistant.setup] Setup failed for zwave: Could not install all requirements.
```

XECDesign, 2017-06-07:
This repo is not related to hassbian. I think you're looking for https://github.com/home-assistant/pi-gen

Vexamus, 2017-06-07:
My bad, thank you!

closed 2017-06-07 by Vexamus