RPi-Distro/pi-gen #82: Force password change on first login

drdelaney, 2017-06-27:
I tried to see if there was already a request for this, but saw none.
I do not know if this will impact a GUI environment, but in a lite environment, it would be perfect.

The passwd command has an option to force the password to be changed on the next login. Is it possible this can be toggled for a fresh image, to help mitigate compromised machines?

as root: passwd -e pi
or edit /etc/shadow, and change the 3rd field to a 0
Example: pi:xxx:0:0:99999:7:::

This, by default, will force the person logging in as pi to change the password, and will not allow the setting of the current password.

drdelaney, 2017-06-27:
I have not tried this, but I am guessing you can append
passwd -e pi
passwd -e root
to the stage1/01-sys-tweaks/00-run.sh script after the chpasswd lines before EOF.

XECDesign, 2017-06-27:
In general, that's a good idea. It was discussed earlier and we think that in a classroom environment, forcing a password change isn't really going to work.

closed 2017-06-27 by XECDesign