RPi-Distro/pi-gen #83: Systemd services

timcoote, 2017-07-10:
This may be irrelevant/too hard/wrong approach.

Some of my Pi provisioning mechanisms interact with running services to confirm that updates have worked as expected. The approach requires systemd within the container.  Is this an approach that is a/ useful in general, b/ likely to be supported?

tia
Tim

XECDesign, 2017-07-10:
Depends on how much complexity it would introduce. I have no intention of doing it, but if somebody sends a sensible PR that's easy to understand and doesn't cause other problems, I am all for it.

timcoote, 2017-07-11:
I think that it would add too much complexity. I tried it briefly, based on this redhat work: http://bit.ly/2ud7PtG, and then found this write up of most of the issues on an Ubuntu based image: http://bit.ly/2udugPp.

Docker's design doesn't suit this usecase well, a VM would be more appropriate. Otoh, I prefer the current design, using docker as it enforces a cleaner configuration process (ie not the way that I've done it ;-) )

closed 2017-07-11 by timcoote

XECDesign, 2017-07-11:
I intend to move from plain chroot to systemd-nspawn. AFAIK, it should be able to make what you want to do easier, but I haven't looked too much into it.