RPi-Distro/pi-gen #88: Building an image with Stretch

guysoft, 2017-07-19:
Hey,
Is there a way to make pi-gen build Stretch?

Thanks

nickfox-taterli, 2017-07-22:
no,raspbian source list and raspi list must be modify too,and more packages name are differnent.

guysoft, 2017-07-22:
Ok, is there a location where this is discussed?
Is there a new branch planned or something like that?

nickfox-taterli, 2017-07-22:
I was build Stretch at 3 hours ago,but something are not stable, modify file patch : https://github.com/nickfox-taterli/pi-gen-chinese/commit/250203562db756bad53f65eb4f6cb1a11c4efd7a

guysoft, 2017-07-22:
Is https://github.com/nickfox-taterli/pi-gen-chinese a fork?
Or its a repo you generated?

nickfox-taterli, 2017-07-22:
I made a lot of changes to fit the CN GFW network environment only, so I created a new repo. But pi-gen features did not change much.

XECDesign, 2017-08-17:
Should be good now.

closed 2017-08-17 by XECDesign

guysoft, 2017-08-17:
I see the move was done on https://github.com/RPi-Distro/pi-gen/commit/8f361a1c38f0cc00d7a04486a93b2fe29f67ee5c

You pushed this from an internal repo?

XECDesign, 2017-08-17:
yes

guysoft, 2017-08-17:
So you committed the stuff on Jul 19, 23 days before this issue was opened, but pushed it only yesterday?
We could have been testing this for a month.

XECDesign, 2017-08-17:
Pretty sure there has already been a discussion about releasing stuff before it's ready.

guysoft, 2017-08-17:
I've been talking about this here for a while with jamesh:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=187797&p=1183905#p1183905
I would value your input, I don't want to step on someones feet by mistake. There is an understandable clash of interests here I don't want it to be taken the wrong way.

XECDesign, 2017-08-17:
posted a reply (as shiftplusone)