RPi-Distro/pi-gen #90: xz: command not found

karloscampus, 2017-07-25:
I try to compile Raspbian lite (SKIP Stage3-5) with the build-docker.sh, but at the end the Script aborts with following error:
`21:15:31] End /pi-gen/export-noobs/prerun.sh
[21:15:31] Begin /pi-gen/export-noobs/00-release
[21:15:31] Begin /pi-gen/export-noobs/00-release/00-run.sh
'files/partition_setup.sh' -> '/pi-gen/work/2017-07-25-MyRaspbian/export-noobs/2017-07-25-MyRaspbian-lite/partition_setup.sh'
'files/partitions.json' -> '/pi-gen/work/2017-07-25-MyRaspbian/export-noobs/2017-07-25-MyRaspbian-lite/partitions.json'
'files/os.json' -> '/pi-gen/work/2017-07-25-MyRaspbian/export-noobs/2017-07-25-MyRaspbian-lite/os.json'
'files/OS.png' -> '/pi-gen/work/2017-07-25-MyRaspbian/export-noobs/2017-07-25-MyRaspbian-lite/OS.png'
'files/release_notes.txt' -> '/pi-gen/work/2017-07-25-MyRaspbian/export-noobs/2017-07-25-MyRaspbian-lite/release_notes.txt'
./
./slides_vga/
./slides_vga/F.png
./slides_vga/C.png
./slides_vga/A.png
./slides_vga/G.png
./slides_vga/D.png
./slides_vga/B.png
./slides_vga/E.png
./00-run.sh: line 13: xz: command not found
./00-run.sh: line 14: xz: command not found
expr: syntax error
Done! Your image(s) should be in deploy/
`

XECDesign, 2017-07-28:
@cfstras I am guessing Dockerfile needs `pxz` added?

XECDesign, 2017-07-28:
Scratch that, it's already there. Hmm

closed 2017-07-28 by XECDesign

cfstras, 2017-07-31:
Seems I missed that... I only ever built stage2.

XECDesign, 2017-07-31:
I didn't have it listed as a dependency anywhere, so this was a problem before Dockerfile was added.

commit 68cdf34c (2017-07-28): Add xz-utils dependency

Closes #90