RPi-Distro/pi-gen #93: Feature Request: Modular Builds

duerig, 2017-08-17:
I've been working on a build script for my own images, similar to the user in #50. While deleting and replacing later stages works to an extent, it is hard to maintain, especially when pulling in new changes from the main repository. But I found that this could be made much easier with just a few minor changes to build.sh:

(1) Make a STAGE_DIRS variable which allows a space-separated list of stages instead of just using stage*. It would of course default to stage*. This would allow people who wanted custom stages to create separate directories for them and so not interfere with the default stage*. It would avoid building unwanted stages and make merging changes from upstream easier.

(2) Change the prerun.sh files to use a new 'copy_stage' that explicitly names the previous stage rather than 'copy_previous'. This would let users just rebuild later stages incrementally when doing development without needing to re-run earlier stages. copy_previous only works  by remembering the stages as they are run. This is the only piece of state that is kept from stage to stage so remove this one thing and incremental builds are easy.

With these two minor changes, you can make it very easy for somebody to check out the repo, create a 'stage-foobar' directory, add that and it's dependant stages to the config file, and start building a new image based on any one of the canonical stages. Then when they want to update their repo later to accept changes made here, it is a simple pull request that requires no deconflicting. And after the first run, they can tweak their config file to only re-run their own stage in future builds while they develop it to improve turnaround time for debugging.

I've got this working on my tree. If you are interested in doing this, I am happy to send a diff or pull request.

XECDesign, 2017-08-24:
Sounds like an improvement. A PR would be welcome.

jacen92, 2017-08-27:
Hello,,

Could we imagine a minimal desktop image? It should simplify the feature if we just have to set the last stage to build.

RichardBronosky, 2017-10-04:
"rebuild later stages incrementally when doing development without needing to re-run earlier stages" I'm trying to do this right now. Any suggestions on how to do this would be appreciated. How are you doing this now?

duerig, 2017-10-04:
I'll try to get my PR ready tonight. Sorry for not following up earlier.
-Jonathon Duerig

Links:

  1. https://github.com/RPi-Distro/pi-gen/issues/93#issuecomment-334156505
  2. https://github.com/

RichardBronosky, 2017-10-04:
Relevant: https://github.com/RPi-Distro/pi-gen/issues/50#issuecomment-334155480

mickeprag, 2017-10-31:
This sounds like a really good idea. +1 from me for this.

bjpirt, 2018-01-15:
Just a quick note for those following this issue; I'm not sure any pull request is necessary because
 - a) you can already skip stages using the SKIP files (see main README)
 - b) you can isolate your changes to your own stage directory. I use stage2-0 which inserts my stage directory neatly between stages 2 and 3 and makes it simple to update from upstream.

or am I missing some use case?

david-kohen, 2018-04-26:
Agree with bjpirt - I use stage2a and just drop it in place

matthijskooijman, 2019-05-12:
I believe this issue can be closed.

Suggestion 1) is implemented now with the `STAGE_LIST` variable. Suggestion 2) is not, but currently the PREV variables include all stages, even ones that have a `SKIP` file (I presume this was not the case when this issue was opened). This is probably even a better solution than the proposal in this issue, since now you can omit a stage from `STAGE_LIST` and the rootfs copying still works (which would break when stages would hardcode the previous stage to copy from).