RPi-Distro/pi-gen #98: Hash Sum Mismatch

Immortalin, 2017-08-25:
Trace
```Shell
E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-doc_1.20170811-1_armhf.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:937d079fbb87a5a01233ad36f3050d7d54b7a9ac0d7705527ae2abe3421a7bbe
    - SHA1:dd554fb85e92ecfeae70b0c716f32bdca83a24fc [weak]
    - MD5Sum:039399ae469f9bc8fcaa080942cc6d09 [weak]
    - Filesize:31405478 [weak]
   Hashes of received file:
    - SHA256:5242fecdf45084a7bc66fd74bae0fe90a5125c3245119edf4744dbb3e38e234b
    - SHA1:04221c05d30d08d0f9a0fef9563a5e7916a53b22 [weak]
    - MD5Sum:a5f5c2addc0b56920f4f873ff1be4de8 [weak]
    - Filesize:4997120 [weak]
   Last modification reported: Thu, 24 Aug 2017 11:33:29 +0000
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Problem occurs when building on both Raspberry Pi running raspbian and x64 Ubuntu machine

malaire, 2017-08-30:
Seems like just a temporary problem with archive.raspberrypi.org or just network error (only partial file downloaded).

Do you still get this error?

XECDesign, 2017-09-05:
No reply

closed 2017-09-05 by XECDesign