RPi-Distro/pi-gen #99: Docker build throws warnings about Term/ReadLine

jcwren, 2017-08-27:
I've just started building images using pi-gen under Docker, and get the warning below. I'm unclear if this is actually a concern or not. If it can safely be ignored, perhaps the README.md could be updated to indicate that.

If it is of concern, what additional information would be needed?

```
debconf: unable to initialize frontend: Dialog
debconf: (TERM is not set, so the dialog frontend is not usable.)
debconf: falling back to frontend: Readline
debconf: unable to initialize frontend: Readline
debconf: (Can't locate Term/ReadLine.pm in @INC (you may need to install the Term::ReadLine module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.24.1 /usr/local/share/perl/5.24.1 /usr/lib/arm-linux-gnueabihf/perl5/5.24 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.24 /usr/share/perl/5.24 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base .) at /usr/share/perl5/Debconf/FrontEnd/Readline.pm line 7, <> line 219.)
debconf: falling back to frontend: Teletype
dpkg-preconfigure: unable to re-open stdin: 
```

jacen92, 2017-08-27:
Hi,

me pull request will fix this (see the Dockerfile, DEBIAN_INTERACTIVE stuff)

XECDesign, 2017-08-31:
Is this fixed by https://github.com/RPi-Distro/pi-gen/commit/05b31b287d54eaf6269a233997eb97770ec7eed3 ?

jacen92, 2017-08-31:
> Is this fixed by 05b31b2

Indeed

XECDesign, 2017-08-31:
@jacen92 Thank you

closed 2017-08-31 by XECDesign