RPi-Distro/pi-gen #102: Build fails often at "Backing up old config files..."

malaire, 2017-08-30:
Building often fails after following messages:

Setting up geany-common (1.29-1) ...
Setting up libgstreamer0.10-0:armhf (0.10.36-1.5) ...
Setting up python-picraft (1.0) ...
Setting up libsdl1.2debian:armhf (1.2.15+dfsg1-4) ...
Setting up python3-spidev (2.0~git20150907+rpi1) ...
Setting up giblib1:armhf (1.2.4-11) ...
Setting up libtxc-dxtn-s2tc:armhf (1.0+git20151227-2) ...
update-alternatives: using /usr/lib/arm-linux-gnueabihf/s2tc/libtxc_dxtn.so to provide /usr/lib/arm-linux-gnueabihf/libtxc_dxtn.so (libtxc-dxtn-arm-linux-gnueabihf) in auto mode
Setting up pypy-lib:armhf (5.6.0+dfsg-4+rpi1) ...
Setting up python-colorama (0.3.7-1) ...
Setting up libqt5scintilla2-l10n (2.9.3+dfsg-4) ...
Setting up raspberrypi-ui-mods (1.20170725) ...
update-alternatives: using /usr/bin/startlxde-pi to provide /usr/bin/x-session-manager (x-session-manager) in auto mode
update-alternatives: using /usr/share/libgksu/debian/gconf-defaults.libgksu-sudo to provide /usr/share/gconf/defaults/10_libgksu (libgksu-gconf-defaults) in manual mode
Backing up old config files...
Backing up old config files...

Sometimes there is following message after "Backing up old config files...", but often there is nothing else, build just hangs there indefinitely:

(process:3536): dconf-WARNING **: failed to commit changes to dconf: Error spawning command line 'dbus-launch --autolaunch=6c0ae75983574faea5b1921152e7df64 --binary-syntax --close-stderr': Child process exited with code 1

I have not been able to determine any clear reason when this happens and when not - seems to be quite random.

Example commands which sometimes trigger this:

    git clone https://github.com/RPi-Distro/pi-gen.git
    cd pi-gen
    echo "IMG_NAME='MyRaspbian'" > config
    time sudo ./build.sh |& tee mylog

XECDesign, 2017-08-31:
I haven't seen this, but will keep an eye out. It may help to see the process tree when this happens.

malaire, 2017-08-31:
Looking log more carefully, this seems to happen during "stage4/00-install-packages/00-packages".

Process tree when build halts (build was started with command `time sudo ./build.sh |& tee mylog`, no warning message about dconf this time):

    sudo───build.sh───bash─┬─apt-get─┬─dpkg─┬─raspberrypi-ui-─┬─gsettings─┬─{dconf worker}
                           │         │      │                 │           └─{gsettings}
                           │         │      │                 └─{raspberrypi-ui-}
                           │         │      └─{dpkg}
                           │         └─{apt-get}
                           └─{bash}

malaire, 2017-09-01:
btw, is there a up-to-date for `raspberrypi-ui-mods`?

Depository https://github.com/RPi-Distro/raspberrypi-ui-mods does NOT contain up-to-date sources, for example file `debian/raspberrypi-ui-mods.postinst` where this error seems to happen is completely different to what is included in mirror.

malaire, 2017-09-01:
I'll add new comments at raspberrypi-ui-mods issue linked above which seems more suitable.

XECDesign, 2017-10-27:
Can't look into it since I'm not seeing it. It happens at a level that has nothing to do with pi-gen or ui-mods, so I suspect there may be a local issue with your setup.

closed 2017-11-04 by XECDesign