RPi-Distro/pi-gen #104: running out of loopback devices, then rsync issues.

timcoote, 2017-09-01:
I'm running pi-gen's build-docker.sh in various fedora vms. I find that I need to create extra loop devices on the guest vm, otherwise docker containers seem to run out in `export-image/prerun.sh`, near here:
`BOOT_DEV=$(losetup --show -f -o ${BOOT_OFFSET} --sizelimit ${BOOT_LENGTH} ${IMG_FILE})
ROOT_DEV=$(losetup --show -f -o ${ROOT_OFFSET} --sizelimit ${ROOT_LENGTH} ${IMG_FILE})
`
like this:
```
+ ROOT_LENGTH=2988912640
++ losetup --show -f -o 4194304 --sizelimit 43854848 /pi-gen/work/2017-09-01-osx/export-image/2017-09-01-osx-4GB.img
losetup: /pi-gen/work/2017-09-01-osx/export-image/2017-09-01-osx-4GB.img: failed to set up loop device: No such file or directory
+ BOOT_DEV=
```
Clearly there's a workaround, eg:
```
# create a couple of dummy files:
dd if=/dev/zero of=virtualfs1 bs=1024 count=30720
dd if=/dev/zero of=virtualfs1 bs=1024 count=30720
# incrementally associate the files with loop devices
sudo losetup -f    # creates loop0
sudo losetup /dev/loop0 virtualfs1
sudo losetup -f   # creates loop1
sudo losetup /dev/loop0 virtualfs2
# remove the associations:
sudo losetup -d /dev/loop0
sudo losetup -d /dev/loop1
```
I don't know whether this is a specific redhat distro issue. I thought it worth pointing out. It would be a pita to automate, I think.

Once the script gets onto the rsync command (`rsync -aHAXx --exclude var/cache/apt/archives ${EXPORT_ROOTFS_DIR}/ ${ROOTFS_DIR}/`), there seems to be some tangling with SELinux as the -X flag (--xattrs) fails for the vfat partitions (/boot and /dev), with errors like this:
```
output.build.osx.7:+ rsync -aHAXx --exclude var/cache/apt/archives /pi-gen/work/2017-09-01-osx/stage4/rootfs/ /pi-gen/work/2017-09-01-osx/export-image/rootfs/
output.build.osx.7:rsync: rsync_xal_set: lsetxattr(""/pi-gen/work/2017-09-01-osx/export-image/rootfs/boot"","security.selinux") failed: Operation not supported (95)
output.build.osx.7:rsync: rsync_xal_set: lsetxattr(""/pi-gen/work/2017-09-01-osx/export-image/rootfs/boot/.COPYING.linux.QQ1rbT"","security.selinux") failed: Operation not supported (95)
```
It looks to me like rsync shouldn't try to transfer attributes to filesystems that cannot handle them, and SELinux isn't yet sufficiently widespread for this issue to occur often.  I'll update this when I find a workaround.

jcwren, 2017-09-02:
I've run into the loop device issue also. It fails every 5th run, so something isn't doing a 'losetup -d /dev/loopx' somewhere. I've been working around it by restarting docker every 3rd build or so.

timcoote, 2017-09-02:
I think that some loop associations don't get removed. They're usually visible with losetup. The issue that I was having occurs immediately with VMs, where there's only 1 loop device when docker starts.

I found that the rsync issue can be avoided by splitting the process into three and avoiding the -X flag where it causes problems:
```
rsync -aHAXx --exclude var/cache/apt/archives --exclude dev --exclude boot ${EXPORT_ROOTFS_DIR}/ ${ROOTFS_DIR}/
rsync -aHAx --exclude var/cache/apt/archives ${EXPORT_ROOTFS_DIR}/boot/ ${ROOTFS_DIR}/boot/
rsync -aHAx --exclude var/cache/apt/archives ${EXPORT_ROOTFS_DIR}/dev/ ${ROOTFS_DIR}/dev/
```

cfstras, 2017-09-04:
@jcwren I've seen this also, and couldn't figure out how to reproduce. I found that umnounting the images (which also happens when deleting the docker container) removes the loop mounts, too, but it seems that cleanup does not work completely.

@timcoote: The rsync issue seems like a second, unrelated problem.

timcoote, 2017-09-04:
@cfstras yup. I was being parsimonious in issue creation ;-) : probably a poor process on my part.

I can reproduce the loop issue by using a fedora host on AWS (ami-5ec82927). Needs 60GB of storage, and I'm using m4-xlarge to get the process to complete quickly. If it's of any use I could create a ready to go ami that reproduces it.

wrt two issues in one, what works from the pov of the repo:
- close both issues as there's a workaround for each
- create and close a separate issue for rsync
- something else (eg leave issue(s) open)

I'm content to do any/none of the above.

XECDesign, 2017-09-04:
In the future, one issue per issue would be ideal. In this case. However, in this case, I think both issues are a wontfix for now.

timcoote, 2017-09-04:
Makes sense.  Both are the result of external dependencies and ought to be fixed upstream, rather than worked around.

Will others stumbling across the issues find the documentation?

closed 2017-09-04 by timcoote

XECDesign, 2017-09-04:
"What documentation?"
Probably not, but we can refer them back to this issue. I suppose we could add a 'Known Issues' section to the README.

ball-hayden, 2017-10-24:
I've just got bitten by this issue (`losetup: /pi-gen/work/2017-10-24-edge/export-image/2017-10-24-edge.img: failed to set up loop device: No such file or directory`) while using the Docker build.

I'm a bit confused by the work around @timcoote suggests - surely creating and destroying loopback devices has no net effect? Certainly, it appears to have no effect on my machine as the build still fails after following that workaround.

I'm wondering if there's an SELinux related issue in creating loopback devices. I've run `setenforce 0` on the host, and that seems to be more successful.

timcoote, 2017-10-24:
Creating the devices raises the high water mark for loop devices. Running the build can consume them. Personally, I only run in vms (fedora), mostly on aws, so I only need to raise the high water mark once.

How did yours fail after the workaround?

ball-hayden, 2017-10-25:
I was seeing the same error @timcoote - `failed to set up loop device: No such file or directory`.

MLeeters, 2017-11-01:
Did someone fix this issue? I see the same issue. Even after removing the device.

We have a VM that runs docker. A NFS share is mounted on that vm.
Which is a shared volume of the docker.

Previously this wasn't a problem, till I restarted the docker because the docker version was updated.

timcoote, 2017-11-01:
I'll be spinning this again tomorrow, to confirm that the workaround still works. It's been working for me with a stock Fedora 26 (ami-5ec82927). I run the commands that I described above as part of the provisioning step, embedded in a Vagrantfile.  I run the process with a clean vm each time, as I found this a reliable approach to avoiding /dev/loop* leaks.

siochs, 2017-12-11:
Having the same Problem here (host: docker -> debian:latest). The workaround porposed by @timcoote  and @jakubvalenta seem to work, but I have got the impression this cannot be the solution to the underlying problem, right?

bjoernhaeuser, 2018-01-16:
Having the same problem right now with docker build (host: Centos 7 -> container is debian)

XECDesign, 2018-11-22:
How would enabling debug output fix a system configuration issue?

cfstras, 2018-11-23:
The actual issue is that loop devices work _some_ of the time, so re-running will fix it in most cases. This seems to be due to the order in which they are allocated and deallocated by docker.

XECDesign, 2018-11-23:
How would I go about reproducing this?

cfstras, 2018-11-26:
Build the image using the docker instructions a few times... My current guess is it happens more often if an intermediate build fails (after the losetup step), and thus the cleanup doesn't happen correctly.