RPi-Distro/pi-gen #105: export-noobs fails

jcwren, 2017-09-02:
Building with the latest version, the NOOBS export appears to be failing. This is not a problem for me (I'm using only the image file), and what I'd actually like to know is can I simply delete the export-noobs directory, or does it need to be disabled in a more proper way?

```
[...]
[16:15:41] Begin /pi-gen/stage3
[16:15:41] End /pi-gen/stage3
[16:15:41] Begin /pi-gen/stage4
[16:15:41] End /pi-gen/stage4
[16:15:41] Begin /pi-gen/stage5
[16:15:41] End /pi-gen/stage5
[16:15:41] Begin /pi-gen/export-image
[16:15:41] Begin /pi-gen/export-image/prerun.sh

Welcome to fdisk (util-linux 2.29.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0x5f5d10cd.

Command (m for help): Created a new DOS disklabel with disk identifier 0xd70b449f.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): 
Using default response p.
Partition number (1-4, default 1): First sector (2048-3891946, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (8192-3891946, default 3891946): 
Created a new partition 1 of type 'Linux' and of size 41.8 MiB.

Command (m for help): Disk /pi-gen/work/2017-09-02-Raspbian/export-image/2017-09-02-Raspbian-lite.img: 1.9 GiB, 1992676864 bytes, 3891947 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xd70b449f

Device                                                                      Boot Start   End Sectors  Size Id Type
/pi-gen/work/2017-09-02-Raspbian/export-image/2017-09-02-Raspbian-lite.img1       8192 93845   85654 41.8M 83 Linux

Command (m for help): Selected partition 1
Partition type (type L to list all types): Changed type of partition 'Linux' to 'W95 FAT32 (LBA)'.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): 
Using default response p.
Partition number (2-4, default 2): First sector (2048-3891946, default 2048): 
Sector 8192 is already allocated.
First sector (93846-3891946, default 94208): Last sector, +sectors or +size{K,M,G,T,P} (94208-3891946, default 3891946): 
Created a new partition 2 of type 'Linux' and of size 1.8 GiB.

Command (m for help): Disk /pi-gen/work/2017-09-02-Raspbian/export-image/2017-09-02-Raspbian-lite.img: 1.9 GiB, 1992676864 bytes, 3891947 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xd70b449f

Device                                                                      Boot Start     End Sectors  Size Id Type
/pi-gen/work/2017-09-02-Raspbian/export-image/2017-09-02-Raspbian-lite.img1       8192   93845   85654 41.8M  c W95 FAT32 (LBA)
/pi-gen/work/2017-09-02-Raspbian/export-image/2017-09-02-Raspbian-lite.img2      94208 3891946 3797739  1.8G 83 Linux

Command (m for help): The partition table has been altered.
Syncing disks.

/boot: offset 4194304, length 43854848
/:     offset 48234496, length 1944442368
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
[... 115 lines cut ...]
Partition number (2-4, default 2): First sector (2048-1638399, default 2048): 
Sector 8192 is already allocated.
First sector (8193-1638399, default 10240): Last sector, +sectors or +size{K,M,G,T,P} (10240-1638399, default 1638399): 
Created a new partition 2 of type 'Linux' and of size 795 MiB.

Command (m for help): Disk /pi-gen/work/2017-09-02-Raspbian/export-image/2017-09-02-Raspbian.img: 800 MiB, 838860800 bytes, 1638400 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x29bcab64

Device                                                                 Boot Start     End Sectors  Size Id Type
/pi-gen/work/2017-09-02-Raspbian/export-image/2017-09-02-Raspbian.img1       8192    8192       1  512B  c W95 FAT32 (LBA)
/pi-gen/work/2017-09-02-Raspbian/export-image/2017-09-02-Raspbian.img2      10240 1638399 1628160  795M 83 Linux

Command (m for help): The partition table has been altered.
Syncing disks.

/boot: offset 4194304, length 512
/:     offset 5242880, length 833617920
mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
mkdosfs: Too few blocks for viable filesystem
Done! Your image(s) should be in deploy/
mbpro:pi-gen xxx$```

jcwren, 2017-09-02:
Could this be failing because the README.md example (repeated below) does not remove the EXPORT* files from stage5?

```
# Example for building a lite system
echo "IMG_NAME='Raspbian'" > config
touch ./stage3/SKIP ./stage4/SKIP ./stage5/SKIP
rm stage4/EXPORT*
sudo ./build.sh  # or ./build-docker.sh
```

XECDesign, 2017-09-03:
Yes, your explanation for the failure makes sense.

closed 2017-09-04 by XECDesign

PhistterbutInc, 2019-08-05:
Another thing I noticed was that using the docker build script there is no folder made called 'work'.

commit 2f032b1d (2017-09-04): Update README.md

Fixes #105