RPi-Distro/pi-gen #115: Install python package using pip while building image

somethingOfChoice, 2017-09-22:
Is there a way to run pip as a bash command to install a python package while generating an image?

Note: We want to install the python package on the image before booting it up, not the host machine.

cfstras, 2017-09-23:
add to any of the run-scripts in the stage folders( after pip has been installed):

    on_chroot <<EOF
    pip install foo
    EOF

somethingOfChoice, 2017-09-27:
Thanks a lot. It worked! Another question, if I wanted to modify the /etc/network/interfaces file and add some custom lines, how would I do that?

XECDesign, 2017-09-27:
Note that the standard image uses dhcpcd, which will not work if you make additions to the interfaces file.

If you must modify it, you can do it as a patch or a file to replace it entirely. I usually use patches where possible because when the upstream file changes, it might not be obvious what the change you made 2 years ago was, what it was for and whether it's still needed.

In this case, it's best to just use a whole file. I store them in a 'files' directory of the relevant substage and then use an install command in a run file to copy it. You should see plenty examples of that.

somethingOfChoice, 2017-09-27:
Can I edit dhcpcd to write to the interfaces file a static ip of 192.168.0.254 for the eth0 interface? I'm building the image in jessie if that helps.

somethingOfChoice, 2017-09-27:
Also, where is the dnsmasq.conf file located in the build? I can't find it.

XECDesign, 2017-09-28:
All the files are in the rootfs directory where they would normally appear, provided you've added the appropriate package.

closed 2017-09-28 by XECDesign