RPi-Distro/pi-gen #116: Package policykit-1 seems to be a dependency for systemd services, but is not installed

guysoft, 2017-09-24:
Hey,
So I am not 100% sure of this, so input would be appreciated, but wanted to report this ASAP.
After upgrading to Stretch several systemctl services started failing with the message
```
The name org.freedesktop.PolicyKit1 was not provided by any .service files
```
Installing ``policykit-1``, which was not initially shipped in Jessie, solves the problem. But I am not really sure why the distro upgrade now requires it int he first place.
You can see the report in our issue tracker here: https://github.com/guysoft/OctoPi/issues/402

ZanderBrown, 2017-09-24:
PolKit is increasing used in software to provide Windows UAC style administration, you have to remember that most of the packages in Jessie are a 18 months / a couple years old whereas in stretch they are the latest versions with often quite comprehensive PolKit integration

guysoft, 2017-09-24:
It failed also on simple systemctl services created by us (the OctoPrint service, and gencert), but also haproxy supplied. So I think its specific to systemctl. Its not related to changes in the packages themselves.

XECDesign, 2017-09-25:
Not sure that I have any input here. If some service requires policykit, the package should list it as a dependency. Or are you saying that all services, no matter how trivial fail to run without policykit?

guysoft, 2017-09-25:
Yep, option number 2. At least its how it looks here.  At least anything that uses ``update-rc.d [service] defaults``

closed 2017-12-13 by XECDesign

guysoft, 2017-12-13:
Yay! Did you figure out in the end why its aa requirement?

XECDesign, 2017-12-13:
It's needed for piserver to allow reboot/shutdown without 'sudo' in Raspbian Lite. Since we need it either way, I didn't look into it.

commit 13185c82 (2017-12-13): stage2: install policykit-1

Closes #116