RPi-Distro/pi-gen #117: Error building on NTFS drive

rcuddy, 2017-09-28:
Running ./build.sh on NTFS partition mount results in following error being thrown by scripts/common bootstrap()

rmdir: failed to remove '/media/Data/code/rpi/pi-gen/work/2017-09-28-Rasbian/stage0/rootfs/debootstrap': Directory not empty

Eventually it occurred to me to try moving this checkout to the OS partition and not the shared data parition and it worked beautifully.

Not suggesting this is a bug with build chain, but as I'm incredibly new to many of the things going on in this project I'm not sure if it's an expected issue or maybe worth making note of in the documentation? Maybe people who know what they're doing wouldn't have even expected it to work!

XECDesign, 2017-09-28:
A PR would be welcome.

rcuddy, 2017-09-28:
Sure thing. Just wanted to make sure -I- understood the problem correctly!

rcuddy, 2017-09-28:
https://github.com/RPi-Distro/pi-gen/pull/118

closed 2017-09-28 by rcuddy

dernasherbrezon, 2017-10-21:
Having the same issue while using non-NTFS partition:

rmdir: failed to remove ‘/root/pi-gen/work/2017-10-21-r2cloud/stage0/rootfs/debootstrap’: Directory not empty

/dev/sda1        20G  1.4G   18G   8% /

/dev/sda1 on / type ext4 (rw,relatime,discard,errors=panic,data=ordered)

XECDesign, 2017-10-23:
@dernasherbrezon, are there any error messages from debbootstrap?

If there are still files there after debootstrap finished, it sounds like there's something going wrong earlier in process.