RPi-Distro/pi-gen #122: Stage3 build seems to be up to stage2

ciarancourtney, 2017-10-08:
I've added SKIP and deleted the EXPORT* from stage4 and 5, expecting this to result in a minimal Desktop dist, a la stage3, but the resulting zip in deploy/ is around 350MB (1.7GB img) and when I boot it, it boots to CLI and `lightdm` needs to be installed to boot to GUI. I'm writing the image using win32diskimager

My [build.log](https://pastebin.com/bbFcH9pr) 

Raspberry Pi reference 2017-10-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9f2941954f64d70c0df0de1923c57bf0ef96ceca, stage2

```
Firmware: https://github.com/raspberrypi/firmware/tree/b2420fc150ae4616f5d9ec24bdaedc630586a529
Kernel: https://github.com/raspberrypi/linux/tree/4153f509b449f1c1c816cf124c314975c3daa824
Uname string: Linux version 4.9.41-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1023 SMP Tue Aug 8 16:00:15 BST 2017

Packages:
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                            Version                      Architecture Description
+++-===============================-============================-============-===============================================================================
ii  adduser                         3.115                        all          add and remove users and groups
ii  alsa-utils:armhf                1.1.3-1                      armhf        Utilities for configuring and using ALSA
ii  apt:armhf                       1.4.7                        armhf        commandline package manager
ii  apt-listchanges                 3.10                         all          package change history notification tool
ii  apt-transport-https:armhf       1.4.7                        armhf        https download transport for APT
```

Am I missing something extra that has to be mounted to the SD after writing the img?

XECDesign, 2017-10-09:
Perhaps you also meant to move the EXPORT files from stage 2 to stage 3? The files in deploy are put there by the export stages.

ciarancourtney, 2017-10-10:
@XECDesign OK that got me a bit further, the final zip was ~720MB, but on boot it seems to be missing the entire wlan UI, chromium et al are installed ok. Any ideas?

XECDesign, 2017-10-11:
Sorry, not sure what you mean. Have you added the things that you want to stage 3?

ciarancourtney, 2017-10-19:
Hi @XECDesign I'm not sure what you mean, it doesn't seem to align with the docs. What are the steps to build an image up to and including stage 3 for example?

XECDesign, 2017-10-23:
stage3 is an intermediate stage that I've never actually exported and run, so I can't really comment on that. The idea is that the raspbian images are provided as an example or as a starting point and then you add/remove things as you see fit.

XECDesign, 2017-10-31:
No response

ciarancourtney, 2017-11-01:
I was able to build and run a stage 4 image exactly as per readme, but following the same process for stage 3 does not work, EXPORT file or no

XECDesign, 2017-11-01:
Do you have a fork somewhere I can take a look at? Or are you able to attach a tarball of your stages?

ciarancourtney, 2017-11-02:
Here is the exact commands to reproduce from the current master:

    echo "IMG_NAME='Raspbian'" > config
    touch ./stage4/SKIP ./stage5/SKIP
    cp stage4/EXPORT_IMAGE stage3/
    rm stage4/EXPORT* stage5/EXPORT*
    sudo ./build.sh

The resulting image will have the LXDE desktop lacking wifi/bluetooth configuration UI

XECDesign, 2017-11-03:
The behaviour you're describing is exactly what I would expect, as mentioned in https://github.com/RPi-Distro/pi-gen/issues/122#issuecomment-338561416

closed 2017-11-03 by XECDesign