RPi-Distro/pi-gen #123: Mosquitto

debsahu, 2017-10-11:
Cant install mosquito using hassbian-config, please help. A fresh install of hasbian.

jcwren, 2017-10-11:
Pretty sure you're going to need to report some kind of actual error message and exactly what you're doing before anyone puts any effort into attempting to help you.

debsahu, 2017-10-11:
Relevant lines: 

> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  mosquitto : Depends: libssl1.0.0 (>= 1.0.1) but it is not installable
>              Depends: libwebsockets3 (>= 1.2) but it is not installable
>  mosquitto-clients : Depends: libmosquitto1 (= 1.4.12-0mosquitto1) but it is not going to be installed
> N: Ignoring file 'mosquitto-stretch.list.1' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
> E: Unable to correct problems, you have held broken packages.
>

debsahu, 2017-10-11:
Nevermind, had to build from source. The hassbian-config doesnt work.

closed 2017-10-12 by XECDesign