RPi-Distro/pi-gen #124: No user pi in stage 3?

RichardBronosky, 2017-10-12:
I have created my own stage3 directory with

```
pi=" --owner pi --group pi "

# Setup vim
install        -d                        $pi ${ROOTFS_DIR}/home/pi/.vim/tmp
install -m 644 files/vimrc               $pi ${ROOTFS_DIR}/home/pi/.vim/
install        -d                            ${ROOTFS_DIR}/root/.vim/tmp
install -m 644 files/vimrc                   ${ROOTFS_DIR}/root/.vim/
```

And I get a message saying that `home/pi` doesn't exist. Have I misunderstood the order of operations?

RichardBronosky, 2017-10-12:
Just to be clear, it works for `root`.

RichardBronosky, 2017-10-12:
I figured it out. The reason that `install -m 644 files/vimrc --owner pi --group pi ${ROOTFS_DIR}/home/pi/.vim/` fails with `home/pi` doesn't exist is because `install -d --owner pi --group pi ${ROOTFS_DIR}/home/pi/.vim/tmp` failed to create the directory for this stage. The reason the directory creation is failing is because I don't have a `user: pi` on my host system. Luckily, I do have a user with `id: 1000`. So for me, my workaround is to use `--owner 1000 --group 1000`.

closed 2017-10-12 by RichardBronosky

Landrash, 2017-10-13:
@RichardBronosky Have a look at the `/etc/skel` directory. You should be able to add your files to the base user that's copied on creation of a user.

RichardBronosky, 2017-10-25:
Thanks @Landrash that is the way I should be adding `.bashrc` and `.vimrc`. But for `.ssh/authorized_keys` files probably not.

However, I really prefer to only add stages on top of pi-gen. There pi user gets created at https://github.com/RPi-Distro/pi-gen/blob/58e7e4023eb6f6cd68230fa33a24cdc20c40d2a9/stage1/01-sys-tweaks/00-run.sh So, I'm not yet comfortable with that one.