RPi-Distro/pi-gen #125: Setting locale failed

ajtaggart, 2017-10-18:
constantly getting ---> 

perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").

never used to get this warning any ideas?

XECDesign, 2017-11-01:
You shouldn't be seeing that after the locales package is installed (stage0/01-locale/).

closed 2017-11-06 by XECDesign

kowenhouston, 2018-03-02:
The issue is because you haven't set your base OS locale settings correctly. The debootstrap environment is pulling over the locale settings.
Run these commands outside of the chroot (on the host/base OS):
```
dpkg-reconfigure locales # add the en_GB.UTF-8 locale and set as default
export LANGUAGE=en_GB.UTF-8 
export LANG=en_GB.UTF-8
export LC_ALL=en_GB.UTF-8
locale-gen en_GB.UTF-8
update-locale en_GB.UTF-8
```

This gets rid of any issues for me when building.