RPi-Distro/pi-gen #126: The file /etc/rpi-issue is a treasure and should be on the /boot partition

RichardBronosky, 2017-10-20:
I am struggling with a dying RaspberryPi  deployed 2,300 miles away. In my efforts to avoid an expensive trip to replace the SD card, I'm trying to build a local model (like in Apollo 13). In doing so I stumbled upon https://raspberrypi.stackexchange.com/a/70612/8375 where I discovered that `/etc/rpi-issue` gives me the exact commit hash of the rpi-gen that produced the image. This is great for me. But as you can imagine (and see in the comments to the answer) it would be great if the info in that file were available to OSes that cannot read EXT file systems. I would be willing to produce a PR that cause exports to put a file on `/boot/` with that info and an explanation of how to go from:

    Raspberry Pi reference 2017-04-10
    Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, acc4f56597e75225b93be3e1e20504a40565a9b9, stage2

to https://github.com/RPi-Distro/pi-gen/commit/acc4f56597e75225b93be3e1e20504a40565a9b9 and also look for a release matching the hash in https://github.com/RPi-Distro/pi-gen/releases

Would that be welcomed?

RichardBronosky, 2017-10-20:
BTW, I'm experiencing a `Segmentation fault` when I [try to] do apt-get installs. It seems to be coming from `pycompile` and also `lsb_release`.

XECDesign, 2017-10-23:
The file is already present on /boot as issue.txt

What would I need to do to reproduce the segfault issue?

RichardBronosky, 2017-10-24:
The segfault was on an RPi deployed 2,300 miles away (or else I would have just re-imaged it) and was caused by a bad python2.7 which I ended up fixing by
* downloading the image indicated in /etc/rpi-issue
* mounting it as I documented at https://gist.github.com/RichardBronosky/75b62d838a81a60947aeb3ba4a1191a8
* copying over the result of `sudo find /mnt/ | grep -E 'python2.7|pyshared|pymodules'`

XECDesign, 2017-10-24:
Can this issue be closed?

closed 2017-10-24 by RichardBronosky