RPi-Distro/pi-gen #127: How to add a directory and files into the image?

somethingOfChoice, 2017-10-20:
Hi,

If I wanted to create a directory "test" in the root directory and a file "screen.txt" into it, how do I do that?

Thanks in advance.

somethingOfChoice, 2017-10-20:
I have this so far in a run.sh in stage2:

install -d						${ROOTFS_DIR}/test
install -d						${ROOTFS_DIR}/rpicluster/config
install -m 755 files/screen.txt                         	${ROOTFS_DIR}/rpicluster/config

And in the files folder in stage2, screen.txt is there. But when I build, it doesn't show up in the root directory.

jcwren, 2017-10-21:
Why would you expect it to? You're installing it into /rpicluster/config
directory, not the root directory.

--jc

Save a life; adopt a shelter animal.