RPi-Distro/pi-gen #128: SSH fails sometimes to generate keys, and if it does it will fail forever

guysoft, 2017-10-22:
Got several reports from this in OctoPi distro, reproduced locally in FullPageOS.
We catch it quite often because these are headless distros where users ssh often.
Its completely random and is determiend on first boot.

Indicator is ``/etc/ssh/ssh_host_*`` files are all at 0 bytes in size.

Workaround is to run ``sudo /usr/bin/ssh-keygen -A``. But that is not always possible if there is no keyboard at hand.

Output of /var/log/syslog:

```
Sep  7 15:54:45 fullpageos systemctl[419]: Removed /etc/systemd/system/multi-user.target.wants/regenerate_ssh_host_keys.service.
Sep  7 15:54:45 fullpageos systemd[1]: Reloading.
Sep  7 15:54:46 fullpageos systemd[1]: Started Regenerate SSH host keys.
Sep  7 15:54:46 fullpageos systemd[1]: Starting Turn on SSH if /boot/ssh is present...
Sep  7 15:54:46 fullpageos systemd[1]: Reloading.
Sep  7 15:54:47 fullpageos systemd[1]: Starting OpenBSD Secure Shell server...
Sep  7 15:54:47 fullpageos systemd[1]: ssh.service: Main process exited, code=exited, status=255/n/a
Sep  7 15:54:47 fullpageos systemd[1]: Failed to start OpenBSD Secure Shell server.
Sep  7 15:54:47 fullpageos systemd[1]: ssh.service: Unit entered failed state.
Sep  7 15:54:47 fullpageos systemd[1]: ssh.service: Failed with result 'exit-code'.
Sep  7 15:54:47 fullpageos sh[433]: Job for ssh.service failed because the control process exited with error code.
Sep  7 15:54:47 fullpageos sh[433]: See "systemctl status ssh.service" and "journalctl -xe" for details.
Sep  7 15:54:47 fullpageos sh[433]: invoke-rc.d: initscript ssh, action "start" failed.
Sep  7 15:54:47 fullpageos sh[433]: ● ssh.service - OpenBSD Secure Shell server
Sep  7 15:54:47 fullpageos sh[433]:    Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
Sep  7 15:54:47 fullpageos sh[433]:    Active: failed (Result: exit-code) since Thu 2017-09-07 15:54:47 UTC; 98ms ago
Sep  7 15:54:47 fullpageos sh[433]:   Process: 454 ExecStart=/usr/sbin/sshd -D $SSHD_OPTS (code=exited, status=255)
Sep  7 15:54:47 fullpageos sh[433]:  Main PID: 454 (code=exited, status=255)
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: Starting OpenBSD Secure Shell server...
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos sshd[454]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos sshd[454]: error: Bind to port 22 on :: failed: Address already in use.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos sshd[454]: fatal: Cannot bind any address.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: ssh.service: Main process exited, code=exited, status=255/n/a
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: Failed to start OpenBSD Secure Shell server.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: ssh.service: Unit entered failed state.
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos systemd[1]: ssh.service: Failed with result 'exit-code'.

```

See:

https://github.com/guysoft/OctoPi/issues/424

XECDesign, 2017-10-23:
Are you able to find out what happens with regenerate_ssh_host_keys.service?

The idea is that service is only disabled when ssh-keygen executes successfully, so it should run each time the pi boots until the keys are properly generated.
https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/debian/raspberrypi-sys-mods.regenerate_ssh_host_keys.service#L10

I wonder if it's timing out.

guysoft, 2017-10-23:
Hey,
I hadn't just yet, its kinda challenging because it would mean to run 
``systemctl status regenerate_ssh_host_keys.service`` when ssh fails, and hope ``ssh-keygen -A -v`` would have some output.
It might be a good idea to add a test to the service that indeed ``/etc/ssh/ssh_host_*_key*`` have sizes larger than 0.

A script like this for the service perhaps?

```
/usr/bin/ssh-keygen -A -v
for i in /etc/ssh/ssh_host_*_key*; do
    actualsize=$(wc -c <"$i")
    if [ $actualsize -eq 0 ]; then
        echo size is over 0 bytes
        exit 1
    fi
done
exit 0
``` 

You could also just make it a one-liner and replace the line ``ExecStart=/usr/bin/ssh-keygen -A -v`` with:
```
/usr/bin/ssh-keygen -A -v ; for i in /etc/ssh/ssh_host_*_key*; do actualsize=$(wc -c <"$i") ;if [ $actualsize -eq 0 ]; then echo size is 0 bytes ; exit 1 ; fi ; done ;
```
Will PR this if you like.

XECDesign, 2017-10-27:
http://unofficialpi.org/Distros/FullPageOS/2017-10-22_2017-09-07-fullpageos-stretch-lite-0.8.1.zip

Should this exhibit the problem or does it have the workaround?

XECDesign, 2017-10-27:
Just reproduced the error you posted:
```
Sep  7 15:54:47 fullpageos sh[433]: Sep 07 15:54:47 fullpageos sshd[454]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
```
That's because you have ssh.socket enabled. The host keys have been created correctly.

guysoft, 2017-10-27:
No workaround yet - why is ssh.socket enabled?
What could enable it?

Might explain this too - https://github.com/guysoft/FullPageOS/issues/177

XECDesign, 2017-10-27:
Haven't look into it since this doesn't happen in Raspbian.

I've also noticed that fullpageos does an extra reboot when starting up. Could you point me to where it does this?

guysoft, 2017-10-29:
Ok, seem like https://github.com/guysoft/FullPageOS/issues/177 is not related. However https://github.com/guysoft/OctoPi/issues/424 still is, because that obsolete code was only in FullPageOS and not OctoPi.

XECDesign, 2017-10-30:
Okay to close this one until it's reproduced in Raspbian? I don't believe the problem you're seeing is inherent to pi-gen or the ssh key regen service.

guysoft, 2017-10-30:
Ok, I'll add the workaround here.
Note there were two separate issues, on in FullPageOS, and another OctoPi, the OctoPi is the one to watch out for.

closed 2017-10-30 by guysoft