RPi-Distro/pi-gen #129: apt-get install needs to use DEBIAN_FRONTEND=noninteractive

RichardBronosky, 2017-10-24:
Some packages, like postfix, start asking questions during install.

XECDesign, 2017-10-24:
I would prefer to preseed values rather than let  the package guess them.

RichardBronosky, 2017-10-24:
Do you have any advice on getting office to install during pi-gen? I get https://www.google.com/search?q=%2Fusr%2Fsbin%2Fpostconf%3A%20fatal%3A%20inet_addr_local%5Bgetifaddrs%5D%3A%20getifaddrs%3A%20Address%20family%20not%20supported%20by%20protocol

XECDesign, 2017-10-25:
What are you installing exactly? The stock config installs libreoffice without any issues.

RichardBronosky, 2017-10-25:
Sorry for the confusion. I updated my previous comment.

RichardBronosky, 2017-10-25:
I'm going to close this one with...
It's better to do an install that tries to be interactive in 2 steps.

### 00-run.sh

```
#!/bin/bash -ex

# Preseed local packages
# See: https://wiki.debian.org/en/PackageManagement/Preseed
#      https://coderwall.com/p/lryimq/postfix-silent-install-on-ubuntu
for file in $(ls files/*.deb 2>/dev/null || true); do
on_chroot << EOF
    debconf-set-selections <<<"postfix postfix/mailname string raspberrypi"
    debconf-set-selections <<<"postfix postfix/main_mailer_type string 'Internet Site'"
EOF
done

# vim: et sw=4 ts=4 sts=4 syntax=sh
```

### 01-packages-nr
```
postfix
```

I created https://github.com/RPi-Distro/pi-gen/issues/130 to deal specifically with Postfix.

closed 2017-10-25 by RichardBronosky

XECDesign, 2017-10-26:
Note that you don't have to stick debconf  stuff in a run.sh file. For example:
https://github.com/RPi-Distro/pi-gen/blob/dev/stage2/01-sys-tweaks/00-debconf